raspiblitz/raspiblitz #3500: Option to switch to clearnet in webui to speed up initial blockchian sync

rootzoll, 2022-12-07:
like assumed in #3329 ... often initial sync just behind Tor is slow. Give option to switch (just bitcoind) to additionally clearnet to speed up sync

rootzoll, 2023-11-14:
Maybe this is not needed when we support assumeUTXO .. push further in the future.

rootzoll, 2024-09-29:
Pushing this to v1.12.0 for same reasons like AssumeUTXO - see https://github.com/raspiblitz/raspiblitz/issues/1178#issuecomment-2381257789

It still makes sense to combine this with AssumeUTXO because we had setups in workshops where Tor is blocked or slowed down extremly .. so any option that speeds up blockchain resync is needed for v1.12.0 to reduce the migration pain.

scarolan, 2025-03-15:
Yes please! I'm going on ten days now with a Raspberry Pi 4 still at only 20%.... 😔

Giving the user an option to do their initial sync over clearnet would be a big improvement.