raspiblitz/raspiblitz #3516: Port Forwarding ?

BenjaminbFrank, 2022-12-14:
**New Flash Install v1.8.0 - 
Socks5() connect to ***.onion:8333 failed: host unreachable**

blitzversion: 1.8.0
chainnetwork: bitcoin / main
 

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-12-14 03:09:49 GMT; 9min ago
    Process: 6566 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 6571 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 6572 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 6573 (bitcoind)
      Tasks: 18 (limit: 9353)
        CPU: 1min 34.440s
     CGroup: /system.slice/bitcoind.service
             └─6573 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Dec 14 03:09:19 raspberrypi systemd[1]: Starting Bitcoin daemon...
Dec 14 03:09:49 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Mon 2022-12-12 19:49:22 GMT, ends at Wed 2022-12-14 03:19:45 GMT. --
Dec 14 03:09:19 raspberrypi systemd[1]: Starting Bitcoin daemon...
Dec 14 03:09:49 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-12-14T03:09:49Z 2 block-relay-only anchors will be tried for connections.
2022-12-14T03:09:49Z init message: Starting network threads…
2022-12-14T03:09:49Z DNS seeding disabled
2022-12-14T03:09:49Z net thread start
2022-12-14T03:09:49Z addcon thread start
2022-12-14T03:09:49Z msghand thread start
2022-12-14T03:09:49Z opencon thread start
2022-12-14T03:09:49Z tor: Got service ID 6yhf4h6g2vym7emsvqtf3l4ske4uk3dlk5utdwyoiug5ckefhqqkslid, advertising service ***.onion:8333
2022-12-14T03:09:49Z AddLocal(***.onion:8333,4)
2022-12-14T03:09:49Z Leaving InitialBlockDownload (latching to false)
2022-12-14T03:09:49Z init message: Done loading
2022-12-14T03:10:04Z Imported mempool transactions from disk: 2878 succeeded, 0 failed, 0 expired, 0 already there, 0 waiting for initial broadcast
2022-12-14T03:10:04Z loadblk thread exit
2022-12-14T03:16:28Z New outbound peer connected: version: 70016, blocks=767321, peer=6 (outbound-full-relay)
2022-12-14T03:16:34Z New outbound peer connected: version: 70016, blocks=767321, peer=7 (outbound-full-relay)
2022-12-14T03:16:46Z New outbound peer connected: version: 70016, blocks=767321, peer=8 (outbound-full-relay)
2022-12-14T03:16:52Z New outbound peer connected: version: 70016, blocks=767321, peer=9 (outbound-full-relay)
2022-12-14T03:18:03Z New outbound peer connected: version: 70016, blocks=767321, peer=13 (outbound-full-relay)
2022-12-14T03:18:41Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-12-14T03:18:49Z New outbound peer connected: version: 70016, blocks=767321, peer=14 (outbound-full-relay)

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2022-12-14 03:18:46 GMT; 59s ago
    Process: 78677 ExecStartPre=/home/admin/config.scripts/lnd.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 78931 ExecStart=/usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.conf (code=exited, status=1/FAILURE)
   Main PID: 78931 (code=exited, status=1/FAILURE)
        CPU: 4.950s

Dec 14 03:18:46 raspberrypi systemd[1]: lnd.service: Consumed 4.950s CPU time.
[... 304 lines cut ...]
ln_default_sync_initial_done="0"
ln_default_running="0"
ln_default_error_full=""
ln_default_version="0.15.0-beta"
ln_default_locked="0"
ln_default_ready="0"
btc_default_synced="1"
btc_default_peers="7"
btc_default_sync_initial_done="1"
btc_default_error_full=""
btc_default_sync_percentage="100.00"
btc_default_online="1"
btc_default_mempool_transactions="3076"
btc_default_port="8333"
btc_default_blocks_behind="0"
btc_default_running="1"
btc_default_activated="1"
btc_default_blocks_data_kb="493060364"
btc_default_version="v23.0.0"
btc_default_blocks_verified="767321"
btc_default_sync_progress="0.999991788403774"
btc_default_blocks_headers="767321"
btc_default_sync_initialblockdownload="0"
btc_default_error_short=""
btc_default_ready="1"

rootzoll, 2022-12-14:
LND is on restarting:
```
Dec 14 03:18:46 raspberrypi lnd[78931]: unable to open databases: unable to open graph DB: channel db cannot revert to prior version
```

You seam to have flashed an older image ... please make sure to have the latest image: v1.8.0c
https://github.com/rootzoll/raspiblitz#downloading-the-software

cstenglein, 2023-06-07:
Closing for now, please re-open if it still appears with 1.9.0

closed 2023-06-07 by cstenglein