raspiblitz/raspiblitz #3519: Ride the Lightning - Error 503 Unable to Connect to LND Server

tlindi, 2022-12-14:
my raspiblitz is image installed amd64 v1.8 dev branch. Private IP 192.168.1.xxx. 

Zero manual configuration changes - been only using UI's. Tor not disablet. LND and bitcoin work - funds flow nicely.

Installed (and rebooted and re-installed) RTL but it can't connect LND.

Seems it is trying only with ipv6. My (SOHO) router offers ipv6 to internal network - but I know nothing about ipv6.

Logs as requested: https://l.termbin.com/mesev

tlindi, 2022-12-14:
Disabling ipv6 as decribed here
https://www.itzgeek.com/how-tos/linux/debian/how-to-disable-ipv6-on-debian-9-ubuntu-16-04.html

fixed issue

Now you just handle it correctly for all the users 😉

rootzoll, 2022-12-14:
There are some users using IPv6 .. I dont think deactivating it for all by default will be the solution here - BUT its a good lead to work towards a fix.

rootzoll, 2022-12-14:
On my machine I changed `localhost` to `127.0.0.1` and now its working. Change see here:
[    jq ".nodes[0].Settings.lnServerUrl = \"https://127.0.0.1:${portprefix}6100\"" | \](https://github.com/rootzoll/raspiblitz/blob/21eb83b3c2b05efd6ef7c653c81c3ff74c3d39c6/home.admin/config.scripts/bonus.rtl.sh#L469)

Set to final tesing.

closed 2022-12-15 by rootzoll