raspiblitz/raspiblitz #3524: IP2Tor disable or revive?

rootzoll, 2022-12-15:
The IP2Tor service provided by FULMO is not working correctly and will not be continued. Thats wy the following issues were reported:

- [ ] #2247
- [ ] #3335
- [ ] #3417

There is some commitment by the community to setup a new instance of this service .- but we have to see if that will be ready for v1.9.0 or in that version the IP2Tor gets deactivated for now.

rootzoll, 2022-12-24:
- [ ] Recheck FULMO services ... and also #2773 for services like LNbits & BTCPay

frennkie, 2022-12-26:
My node has not been stable over the last month.. sorry about that.. :-/

Also the backup node (Fulmo Node02) wasn't paired correctly... this has been fixed now.

poweruser0815, 2023-01-14:
HI, don´t now why - but now it is working (Fulmo Node) since approx. 10 days

unlockingscript, 2023-01-15:
I can confirm, I am working with it since 3rd Jan. Works better for me than before.

openoms, 2023-04-13:
the default:
`fulmo7x6yvgz6zs2b2ptduvzwevxmizhq23klkenslt5drxx2physlqd.onion`
is down.
Are there alternatives?

rootzoll, 2023-05-03:
ip2tor.com shoudl become the new default shop ... but its also having some problems - see here: https://github.com/rootzoll/raspiblitz/issues/3417#issuecomment-1533472557

rootzoll, 2023-05-30:
Quick update: tested today the ip2tor.com (that is the first option in the RaspiBlitz v1.9.0 SSH subscription menus) and its now working again.

closed 2023-05-30 by rootzoll