raspiblitz/raspiblitz #3535: LNbits /withdraw not working with front end nginx proxy

tlindi, 2022-12-16:
I read many similar SSL/cert issues but none matched to my issue.

I want to access LNbits LNURL without VPS/VPN and any external vendors except my ISP.

I have configured DNS correctly and hostname matching LetsEncrypt cert is used in all the required locations.

I have tested following setups:

**Working setup 1)**
browser<->private IP-net<->raspiblizt-ngnix(ssl 5001 proxy_pass http 5000 to)<->lnbits

**Optional Working setup 2)**
browser<->internet<->router (with ssl 443 portforward to https)<->raspiblizt-ngnix(ssl 5001 proxy_pass http 5000 to)<->lnbits

**Optional Working setup 3)**
browser<->internet<->router (with ssl 443 portforward to https)<->raspiblitz nginx(ssl 443 proxy_pass https 5001 to)<->raspiblizt-ngnix(ssl 5001 proxy_pass http 5000 to)<->lnbits

**Not working Setup 4) with separate host as front nginx host to proxy 443 to raspiblitz (and other hosts too) <- ISSUE**
browser<->internet<->router (with ssl 443 portforward to https)<->front nginx host (ssl 443 proxy_pass https 5001 to)<->raspiblizt-ngnix(ssl 5001 proxy_pass http 5000 to)<->lnbits

**SETUP 1)**
Please study following logs for working setup here:
[image]

Browser console debug shows no issues and networking seems ok.
[image]

**SETUP 4)**
Not working case logs:
[image]

LNbits shows an error:
[image]

Browser debug shows an error and networking has some issues:
[image]

[image]

**_Only case when I have issue is when nginx proxy is on another host!_**
My setup has many hosts on internal network to where I proxy_pass trafic from this front nginx. So it is mandatory to forward all 443 from internet to nginx outside of raspiblitz. Multible browsers were tested too and cookies / browser data was cleared also. All the other sites and their functionalities work. Event other parts of LNbits itself are ok. Also LND channels are fine and funds flow.

No avail. any ideas

ps. sorry about long issue and bad pictures.

tlindi, 2022-12-17:
issue cleared some samples from interwebs had escape "\" for $ on these nginx lines. Like this
  proxy_set_header Host \$http_host;
  proxy_set_header X-Real-IP \$remote_addr;
  proxy_set_header X-Forwarded-For \$proxy_add_x_forwarded_for;
  
 Removed "\" as following
   proxy_set_header Host $http_host;
  proxy_set_header X-Real-IP $remote_addr;
  proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
  
  and now all good :)

closed 2022-12-20 by tlindi

rootzoll, 2022-12-21:
@tlindi you closed this issue .. was there anything in there to be changed on the raspiblitz code side?

tlindi, 2022-12-21:
No need to make changes on raspiblitz - I had escape "\" typo on my config.