raspiblitz/raspiblitz #3536: Self-Built Installation PIP Issue

andruno86, 2022-12-17:
Just ran the script after a fresh install of the RPI image. 

Installer script stops / crashes at the following:

**_"ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    packaging==21.3 from https://www.piwheels.org/simple/packaging/packaging-21.3-py3-none-any.whl#sha256=36bb3108f8e33f88e52e7985b4068e2585b67e5b7b3a0fa2817f9852bc39d2ef (from -r requirements.txt (line 498)):
        Expected sha256 dd47c42927d89ab911e606518907cc2d3a1f38bbd026385970643f9c5b8ecfeb
        Expected     or ef103e05f519cdc783ae24ea4e2e0f508a9c99b2d4969652eed6a2e1ea5bd522
             Got        36bb3108f8e33f88e52e7985b4068e2585b67e5b7b3a0fa2817f9852bc39d2ef"_**

Anyone know how to get around this?

HiLivin, 2022-12-17:
I confirm the same problem with my build.

andruno86, 2022-12-17:
Hopefully there's an easy fix someone can help with!

HiLivin, 2022-12-17:
Related issue in fusion44/blitz_api#161.
I've managed to install current `blitz_api` locally without issues - it doesn't complain about change in hash of `packaging-21.3`.

andruno86, 2022-12-23:
What's the blitz_api? I'm just trying to install with the headless option selected building from the script. Still keeps stopping/crashing at that hash check.

rootzoll, 2022-12-24:
@HiLivin The v1.9.0n Release Candidate 1 is not out and the branch `v1.9` exists - please try that one for building your sd card. I just tested the builds and it worked on my side. 

@andruno86 the `blitz_api` is the python REST backend needed for the new WebUI. You can build a sdcard without that API when truning off fatpack with parameter `-f 0` on calling the `build_sdcard.sh` script.

cstenglein, 2023-06-07:
@HiLivin  is this still an issue with 1.9.0?

Closing this for now, please re-open if it still persists.

closed 2023-06-07 by cstenglein