raspiblitz/raspiblitz #3548: Stuck on prepare sync of CLN after upgrade

salenjak, 2022-12-19:
Hi,

I have updated Raspiblitz  to v1.8.0c and CLN to v22.11.1.

The sync (Lightning) has been sitting at 100% for hours. 
[image]

Raspiblitz status screen:
[image]

Here is debug log:
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.8.0
chainnetwork: bitcoin / main
 14:46:46 up  2:50,  2 users,  load average: 2.26, 2.50, 2.55

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-12-19 11:58:29 GMT; 2h 48min ago
    Process: 5810 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 5824 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 5826 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 5911 (bitcoind)
      Tasks: 18 (limit: 9353)
        CPU: 28min 57.030s
     CGroup: /system.slice/bitcoind.service
             └─5911 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Dec 19 11:58:09 raspberrypi systemd[1]: Starting Bitcoin daemon...
Dec 19 11:58:29 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Thu 2022-07-28 15:25:25 BST, ends at Mon 2022-12-19 14:46:46 GMT. --
Dec 19 11:58:09 raspberrypi systemd[1]: Starting Bitcoin daemon...
Dec 19 11:58:29 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-12-19T13:28:21Z New outbound peer connected: version: 70016, blocks=768082, peer=66 (outbound-full-relay)
2022-12-19T13:33:19Z UpdateTip: new best=0000000000000000000513d49af89b4a8196261dd613eaf0a1561430208a0ebc height=768083 version=0x20004000 log2_work=93.903494 tx=789297595 date='2022-12-19T13:33:04Z' progress=1.000000 cache=14.9MiB(109935txo)
2022-12-19T13:34:40Z UpdateTip: new best=00000000000000000001c23e7196c1e83720aefb146bab747d828d5dae8a90e9 height=768084 version=0x20600004 log2_work=93.903506 tx=789300129 date='2022-12-19T13:34:18Z' progress=1.000000 cache=15.5MiB(115145txo)
2022-12-19T13:35:04Z UpdateTip: new best=00000000000000000007989a7d1353c6ff3d25eb75cccd66e9ccd3ffb6ebbd4b height=768085 version=0x20200000 log2_work=93.903517 tx=789301306 date='2022-12-19T13:34:39Z' progress=1.000000 cache=16.8MiB(125680txo)
2022-12-19T13:36:52Z UpdateTip: new best=00000000000000000000adadd792b98518f0ab21e27980e2008248fce0210a4a height=768086 version=0x293ac000 log2_work=93.903529 tx=789302058 date='2022-12-19T13:36:33Z' progress=1.000000 cache=17.0MiB(127299txo)
2022-12-19T13:41:01Z New outbound peer connected: version: 70016, blocks=768086, peer=75 (outbound-full-relay)
2022-12-19T13:51:39Z New outbound peer connected: version: 70016, blocks=768086, peer=79 (block-relay-only)
2022-12-19T14:02:49Z New outbound peer connected: version: 70016, blocks=768086, peer=85 (block-relay-only)
2022-12-19T14:07:52Z UpdateTip: new best=00000000000000000003a0fe6c29f3ca331725be1a713a98211a9cdce997bf15 height=768087 version=0x23b32004 log2_work=93.903540 tx=789304837 date='2022-12-19T14:07:41Z' progress=1.000000 cache=19.6MiB(148258txo)
2022-12-19T14:11:30Z UpdateTip: new best=0000000000000000000139d62c2755ceadd35727d2814b6ef61e14160f675a6d height=768088 version=0x37ffe000 log2_work=93.903552 tx=789307293 date='2022-12-19T14:11:01Z' progress=1.000000 cache=20.1MiB(152508txo)
2022-12-19T14:29:21Z UpdateTip: new best=0000000000000000000625ee774c5609cff29c40e7ce66359fb463c46dd5e794 height=768089 version=0x31710004 log2_work=93.903563 tx=789310326 date='2022-12-19T14:28:53Z' progress=1.000000 cache=21.6MiB(164343txo)
2022-12-19T14:30:08Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-12-19T14:31:46Z New outbound peer connected: version: 70016, blocks=768089, peer=102 (outbound-full-relay)
2022-12-19T14:32:04Z Socks5() connect to ***.onion:8333 failed: host unreachable
2022-12-19T14:35:25Z New outbound peer connected: version: 70016, blocks=768089, peer=109 (outbound-full-relay)
2022-12-19T14:35:35Z New outbound peer connected: version: 70016, blocks=768089, peer=110 (outbound-full-relay)
2022-12-19T14:37:19Z New outbound peer connected: version: 70016, blocks=768089, peer=111 (outbound-full-relay)
2022-12-19T14:38:15Z UpdateTip: new best=000000000000000000037fc526c33248f405a52b249c325af9c03c67bbbc55d6 height=768090 version=0x20006000 log2_work=93.903575 tx=789312633 date='2022-12-19T14:37:50Z' progress=1.000000 cache=22.3MiB(170374txo)
2022-12-19T14:40:19Z New outbound peer connected: version: 70016, blocks=768090, peer=113 (block-relay-only)
2022-12-19T14:43:02Z New outbound peer connected: version: 70016, blocks=768090, peer=115 (block-relay-only)
[... 308 lines cut ...]
ln_default_version="v22.11.1"
ln_default_running="0"
ln_default_ready="0"
ln_default_locked="0"
ln_default_activated="1"
ln_default_error_short=""
btc_default_blocks_data_kb="494107372"
btc_default_running="1"
btc_default_error_full=""
btc_default_sync_initial_done="1"
btc_default_sync_initialblockdownload="0"
btc_default_online="1"
btc_default_blocks_headers="768090"
btc_default_port="8333"
btc_default_blocks_behind="0"
btc_default_error_short=""
btc_default_activated="1"
btc_default_synced="1"
btc_default_peers="15"
btc_default_ready="1"
btc_default_mempool_transactions="4877"
btc_default_sync_progress="0.9999980221314662"
btc_default_version="v23.0.0"
btc_default_blocks_verified="768090"
btc_default_sync_percentage="100.00"

salenjak, 2022-12-19:
Also, I have noticed that when running "lightning-cli help" command in terminal I get "lightning-cli: Connecting to 'lightning-rpc': Connection refused".
in WEBGUI there is a next message: 
"Bitcoin done
Lightning offline
Unable to connect to CLN daemon, waiting..."

Does anyone have any idea what is the problem here?

salenjak, 2022-12-20:
Still OFFLINE :(

For som reason lightnind is restarting all the time.
I have started it with "systemctl start lightningd" and on the first check "systemctl status lightningd" it shows that service is active and after few seconds it goes offline (it restarts)

admin@192.168.1.6:~ ₿ systemctl start lightningd
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to start 'lightningd.service'.
Multiple identities can be used for authentication:
 1.  ,,, (pi)
 2.  ,,, (admin)
 3.  ,,, (blitzapi)
 4.  ,,, (joinmarket)
Choose identity to authenticate as (1-4): 2
Password:
==== AUTHENTICATION COMPLETE ===
admin@192.168.1.6:~ ₿ systemctl status lightningd
● lightningd.service - c-lightning daemon on mainnet
     Loaded: loaded (/etc/systemd/system/lightningd.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-12-20 03:35:00 GMT; 3s ago
    Process: 114560 ExecStartPre=/home/admin/config.scripts/cl.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
   Main PID: 114604 (sh)
      Tasks: 29 (limit: 9353)
        CPU: 2.695s
     CGroup: /system.slice/lightningd.service
             ├─114604 /bin/sh -c (cat /dev/shm/.cl.pw;echo;cat /dev/shm/.cl.pw) | /usr/local/bin/lightningd                         --conf=/home/bitcoin/.lightning/config --encrypted-hsm                         --pid-file=/run/lightning>
             ├─114616 /usr/local/bin/lightningd --conf=/home/bitcoin/.lightning/config --encrypted-hsm --pid-file=/run/lightningd/lightningd.pid
             ├─114716 /usr/local/libexec/c-lightning/plugins/autoclean
             ├─114719 /usr/local/libexec/c-lightning/plugins/chanbackup
             ├─114723 /usr/local/libexec/c-lightning/plugins/bcli
             ├─114725 /usr/local/libexec/c-lightning/plugins/commando
             ├─114730 /usr/local/libexec/c-lightning/plugins/fetchinvoice
             ├─114733 /usr/local/libexec/c-lightning/plugins/funder
             ├─114736 /usr/local/libexec/c-lightning/plugins/topology
             ├─114741 /usr/local/libexec/c-lightning/plugins/keysend
             ├─114745 /usr/local/libexec/c-lightning/plugins/offers
             ├─114751 /usr/local/libexec/c-lightning/plugins/pay
             ├─114756 /usr/local/libexec/c-lightning/plugins/txprepare
             ├─114764 /usr/local/libexec/c-lightning/plugins/spenderp
             ├─114769 /usr/local/libexec/c-lightning/plugins/bookkeeper
             ├─114771 /home/bitcoin/cl-plugins-available/sparko
             ├─114776 python3 /home/bitcoin/cl-plugins-available/plugins/backup/backup.py
             └─114786 /home/bitcoin/cl-plugins-available/cln-grpc/debug/cln-grpc

admin@192.168.1.6:~ ₿ systemctl status lightningd
● lightningd.service - c-lightning daemon on mainnet
     Loaded: loaded (/etc/systemd/system/lightningd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2022-12-20 03:35:05 GMT; 26s ago
    Process: 114560 ExecStartPre=/home/admin/config.scripts/cl.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 114604 ExecStart=/bin/sh -c (cat /dev/shm/.cl.pw;echo;cat /dev/shm/.cl.pw) | /usr/local/bin/lightningd                         --conf=/home/bitcoin/.lightning/config --encrypted-hsm                         --pid-file=/run/l>
   Main PID: 114604 (code=exited, status=1/FAILURE)
        CPU: 4.482s

salenjak, 2022-12-20:
There is also some database upgrade error after upgrading:
sudo journalctl -fu lightningd
-- Journal begins at Mon 2022-12-19 21:21:49 GMT. --
Dec 20 04:17:46 raspberrypi systemd[1]: lightningd.service: Consumed 2.100s CPU time.
Dec 20 04:18:17 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 31.
Dec 20 04:18:17 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
Dec 20 04:18:17 raspberrypi systemd[1]: lightningd.service: Consumed 2.100s CPU time.
Dec 20 04:18:17 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Dec 20 04:18:17 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
Dec 20 04:18:19 raspberrypi sh[148314]: Refusing to irreversibly upgrade db from version 197 to 212 in non-final version v22.11.1 (use --database-upgrade=true to override)
Dec 20 04:18:19 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Dec 20 04:18:19 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.

salenjak, 2022-12-20:
I have downgraded CLN and now everything works.

config.scripts/cl.install.sh update v0.11.2rc3

closed 2022-12-20 by salenjak

cintramachado, 2023-08-19:
Hi, Buddy. How to downgrade CLN?