raspiblitz/raspiblitz #3550: Alby will not connect

dskvr, 2022-12-19:
Details:
1. Migrated from Umbrel
2. Have confirmed ISP is not blocking tor
3. Have connected successfully to port 80 over tor
4. Have 100x checked onion and macaroon details.
5. When visiting `hash.onion:8080/v1/getInfo` the page never loads. 

Are there any troubleshooting guides? I have been running in circles but cannot find any relevant details.

openoms, 2022-12-20:
type in terminal:
```
blitzhelp
```
for command line help.
Have a look around in the menu.
Use the command:
```
debug
```
Can share the link from:
```
debug -l
```

hpcodecraft, 2022-12-21:
I just updated to 1.8.0c and have the same issue. I suspect it might be a problem with the SSL certificate.
When I open `hash.onion:8080/v1/getinfo` with Tor browser, it shows a certificate problem.
I then see an error page about the missing macaroon when I continue anyway.

However when I connect using my local LAN IP, it works flawlessly.

openoms, 2023-01-07:
It is a self-signed certificate by LND. Click Advanced and accept it. The borwser will connect afterwards.

dskvr, 2023-01-11:
@openoms https://l.termbin.com/hljm

dskvr, 2023-01-11:
Confused, you asked me to shared it 
> Can share the link from:

openoms, 2023-01-11:
> @openoms Confused, you asked me to shared it.

I meant I am not sure what that tls error in your logs is about.

dskvr, 2023-01-11:
I reset TLS
- still getting connection errors. 
- still not fully loading page on raspiblitz web via tor

here is the new log after TLS reset, and attempting to connect to web via tor 

https://l.termbin.com/pk5tg

_Do you have an lnurl?_

cstenglein, 2023-06-07:
@dskvr  is this still an issue?

Closing this for now, please re-open if it still persists.