raspiblitz/raspiblitz #3555: Services - Custom Restart Lnbits js typo in instructions

tlindi, 2022-12-20:
[image]

restart line says "nlbits"

correct should be "lnbits"

so if user wants to copy paste restart command, it will fail.

rootzoll, 2022-12-20:
I mean you typed in "nlbits", right? typo in, typo out

tlindi, 2022-12-20:
shouldn't script verify that typed service exists before proceeding and and informing user?!?

rootzoll, 2022-12-20:
should is a good question. It means that script needs to be maintained by some developer more constanly - and get extended with every new possible service added. As far as I understood its just a learning tool showing you the real linux command you can use in the future directly. If sombody would like to spend his dev time on maintaining & extending that script in the future thats fine with me (its open source) happy to merge PRs .. but if not I think its also fine as it is.

closed 2022-12-20 by rootzoll