raspiblitz/raspiblitz #3556: Blitz restore Seed+SCP 228 restarts

jessehodl, 2022-12-21:
Hello everyone,

Please find the termbin link here: https://l.termbin.com/r8n5

I am trying to restore my node with my seed and the channel.backup file.
Currently, the node is stuck on 228 restarts. I believe something is off with the bitcoins, but I am not 100% sure, nor can I pinpoint precisely the issue.

There were indications that my disk had started acting up before. It could be now as well. Would I get this far with a broken disk? Can I maybe check on windows to make sure it's alright?

Happy to try anything out, if you guys have feedback regarding the debug logs.

Thanks!

jessehodl, 2022-12-21:
fdisk -l shows my SSD with the correct values. I'd be surprised if the SSD was faulty

jessehodl, 2022-12-21:
I think the issue might be this: 

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-12-21T06:09:08Z Imported mempool transactions from disk: 362 succeeded, 0 failed, 0 expired, 0 already there, 0 waiting for initial broadcast
2022-12-21T06:09:08Z loadblk thread exit
2022-12-21T06:09:13Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1464, nPos=12192414                 0)
2022-12-21T06:09:13Z *** ThreadSync: Failed to read block 0000000000000000000767395f6cb868e7292adff8d0f392697bac8fe177bb0a from disk
2022-12-21T06:09:13Z Error: A fatal internal error occurred, see debug.log for details
2022-12-21T06:09:13Z txindex thread exit
2022-12-21T06:09:13Z tor: Thread interrupt
2022-12-21T06:09:13Z Shutdown: In progress...
2022-12-21T06:09:13Z addcon thread exit
2022-12-21T06:09:13Z RemoveLocal(***.onion:8333)
2022-12-21T06:09:13Z opencon thread exit
2022-12-21T06:09:13Z torcontrol thread exit
2022-12-21T06:09:13Z net thread exit
2022-12-21T06:09:13Z msghand thread exit
2022-12-21T06:09:13Z DumpAnchors: Flush 0 outbound block-relay-only peer addresses to anchors.dat started
2022-12-21T06:09:13Z DumpAnchors: Flush 0 outbound block-relay-only peer addresses to anchors.dat completed (0.00s)
2022-12-21T06:09:13Z scheduler thread exit
2022-12-21T06:09:13Z Writing 0 unbroadcast transactions to disk.
2022-12-21T06:09:13Z Dumped mempool: 0.000394s to copy, 0.013367s to dump
2022-12-21T06:09:14Z Shutdown: done

openoms, 2022-12-21:
you will need to resync the chain as data is corrupted.
Type:
```
menu
```
and can reset the blockchain in the REPAIR menu.

jessehodl, 2022-12-21:
Hey @openoms 
Thanks for the reply!

Okay. I am currently in the process of restoring it. I downloaded the chain, and I am copying it over. I recently reset the pi. Should I just try your route again, or do I let the Blockchain data sync directly, which will take forever, but is worth a shot?

openoms, 2022-12-21:
currently your chain is not good either because of a disk issue or bad source if you copied (the source node must be stopped).

You can also try to recover with Blixt wallet I think.

jessehodl, 2022-12-21:
Okay, I will try and copy it again and see. If that doesn't work, I will buy a disk and try that. 
In the meanwhile I will try Blixt and see what that is all about