raspiblitz/raspiblitz #3579: nginx: [emerg] cannot load certificate

ingo1111, 2022-12-28:
Hello
After a system upgrade, LnBits, the raspiblitz webui and certainly other services no longer work.
RTL works without problems and receiving and sending also works.

DEBUG OUTPUT: https://l.termbin.com/9pgw

output of :

      ₿ sudo nginx -t
      nginx: [emerg] cannot load certificate "/mnt/hdd/app-data/nginx/tls.cert": BIO_new_file() failed (SSL: error:02001002:system 
      library:fopen:No such file or directory:fopen('/mnt/hdd/app-data/nginx/tls.cert','r') error:2006D080:BIO routines:BIO_new_file:no such file)
      nginx: configuration file /etc/nginx/nginx.conf test failed

An "apt update" and "apt upgrade" was aborted with an error message: https://pastebin.com/rsY3GpLv

Using "sudo apt-get install -f" as well as "sudo dpkg --configure -a" prints the same error.

The file "fullchain.cer" is actually not located in the linked directory. 

I suspect the error is this missing certificate.
Unfortunately, my knowledge of servers is slightly limited and I can not solve the problem without help.

warm greetings

geco91, 2022-12-28:
Have you already tried to flash a new sd card and let the system recover? You can use the new v1.9.0 Release Candidate image for this - see: https://github.com/rootzoll/raspiblitz/tree/v1.9#downloading-the-software

ingo1111, 2022-12-28:
Yes. I performed a system restore with version 1.8.0c. Unfortunately it brought the same result.
I will gladly try a rescue with the RC 1.9.0.

ingo1111, 2022-12-29:
Restoration with the version 1.9.0 ran without errors, but the problem was not fixed.

https://l.termbin.com/os60

rootzoll, 2023-01-18:
Did you just ran a "recover" or did you tried the LND rescue file import in a fresh setup?

ingo1111, 2023-01-20:
Hello
I caused this error and it was not caused by the Raspiblitz software. 

Explanation:
I suspect that the loss of the certificate was caused by a previous malfunction. 
Because, after an ordinary LND update (to version 0.15.5-beta) , the SSD was not mounted after the reboot.
Unfortunately I was too lazy to fix this circumstance and updated to version 1.8.0c instead. After that the link to the "fullchain.cer" was broken.
Actually I should have only checked and changed the UUID in the fstab. Then I would not have had this problem now. And this is a really big problem.

As I wrote in the beginning, I don't have enough knowledge about the behavior of certificates and web servers.
I will have another look at the problem this weekend and if I don't find a solution, I will close all channels and reinstall the Raspiblitz.

Translated with www.DeepL.com/Translator (free version)

closed 2023-01-29 by rootzoll