raspiblitz/raspiblitz #3580: Implementation of a ScreenSaver/display OFF function for more privacy (coding needed)

Nodetar, 2022-12-28:
Dear Raspiblitz Community

I have thought about the "ScreenSaver/Display Off" function again because the "hdmi" or "headless" commands are not a real solution and the node reboots every time.

Since in the past people often asked if it is possible to deactivate the info display or the display so that it does not show all the details all the time, I have now come to a privacy solution. However, I am not a coder/programmer and so I hope that you yourself or someone else from the Raspiblitz community can possibly implement this.

It would be good to use a simple command like "dlock" in the console (after you are logged in) to create a ScreenSaver or "Lock Screen". This can look like the blue screen (without "bitcoin" in top bar) at the beginning where you have to enter your Passowrt C for the LND Wallet to unlock it.

In other words, if the node is up and synced, you can go into the console with ctrl+c, enter the command "dlock" and get a blue display with a small gray field where it says "Unlock me if you can..." and below that an "OK" field. If you confirm this "OK" field in the terminal or by touch screen you will get a password input window. As soon as you enter the password, the normal Info Dispay is displayed again.

I hope that this is not too difficult to implement, because it simply provides more privacy, should the node be in someone's living room, for example, because there is also the router and it is not made directly obvious to every visitor, what this little part is there, where something with Bitcoin & Co. is written on the display.

I would be very grateful for feedback and would be happy if someone could program the whole thing.

Regards,
Nodetar

ChuckNorrison, 2022-12-29:
For privacy reasons, better use the new webui and no lcd display on the pi.

Unlock with a password would be bad if no input device is available.

Nodetar, 2022-12-29:
Of course, the ScreenSaver should only be used if you can unlock it again with a keyboard or via an ssh app.

ChuckNorrison, 2022-12-29:
If you already use an input device and want to disable the lcd screen, try to switch session with CTRL + ALT + F3 or simply CTRL + C.

Nodetar, 2023-01-02:
None of the proposals achieve what I am trying to implement or suggest here. Still would appreciate a ScreenSaver function.

rootzoll, 2023-01-18:
I think for everything concerning the functions of the LCD screen it would make sense to wait until we have a revive of the TUI (Touchscreen Interface) - there you can then have a screensaver and  even a simple PIN number entry to unlock. The LCD is a touchscreen actually - not a good one, but fat finger inputs should work good enough for such use cases. The only thing as that the TUI might take a while - before it gets reworked.