raspiblitz/raspiblitz #3584: Copy Blockchain (sftp)

muraschal, 2022-12-30:
I think there is an error in the sftp markup:
`sftp -r ./chainstate ./blocks bitcoin@192.168.1.112:/mnt/hdd/bitcoin
`
[image]

As you can see, I can connect to filezilla without any problems:
[image]

[image]

muraschal, 2022-12-30:
Okay ich konnte mir selber helfen, mit scp klappts.
Warum in der Anleitung aber sftp deklariert wurde verstehe ich nicht.
:)

Surenic, 2023-01-02:
same here, used scp instead of sftp to copy the blockchain

rootzoll, 2023-01-18:
Thx for the report. Will set back to scp for windows blockchain upload for v1.9.0rc2

closed 2023-01-18 by rootzoll