raspiblitz/raspiblitz #3585: Issue launching LNBits

VeniVidiViciLA, 2023-01-02:
Hi guys,

I have a node that has been up for couple of years now and today I wanted to add the LNBits service. It installed just fine but I'm getting the following error message during the service launch:

Jan 02 21:53:03 raspberrypi systemd[1]: Started lnbits.
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.57 | INFO | Starting LNbits
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.57 | INFO | Host: 127.0.0.1
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.57 | INFO | Port: 5000
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.57 | INFO | Debug: False
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.57 | INFO | Site title: LNbits
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.58 | INFO | Funding source: LndRestWallet
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.58 | INFO | Database: SQLite
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.58 | INFO | Data folder: /mnt/hdd/app-data/LNBits
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.58 | INFO | Git version: 3b98a10c9e65ea35e4e0fc3ff4f417954860f13a
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.58 | INFO | Started server process [55199]
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.58 | INFO | Waiting for application startup.
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.65 | ERROR | Traceback (most recent call last):
Jan 02 21:53:08 raspberrypi sh[55199]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 635, in lifespan
Jan 02 21:53:08 raspberrypi sh[55199]:     async with self.lifespan_context(app):
Jan 02 21:53:08 raspberrypi sh[55199]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 530, in __aenter__
Jan 02 21:53:08 raspberrypi sh[55199]:     await self._router.startup()
Jan 02 21:53:08 raspberrypi sh[55199]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 612, in startup
Jan 02 21:53:08 raspberrypi sh[55199]:     await handler()
Jan 02 21:53:08 raspberrypi sh[55199]:   File "./lnbits/app.py", line 108, in check_wallet_status
Jan 02 21:53:08 raspberrypi sh[55199]:     logger.error(
Jan 02 21:53:08 raspberrypi sh[55199]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/loguru/_logger.py", line 1989, in error
Jan 02 21:53:08 raspberrypi sh[55199]:     __self._log("ERROR", None, False, __self._options, __message, args, kwargs)
Jan 02 21:53:08 raspberrypi sh[55199]:   File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/loguru/_logger.py", line 1954, in _log
Jan 02 21:53:08 raspberrypi sh[55199]:     log_record["message"] = message.format(*args, **kwargs)
Jan 02 21:53:08 raspberrypi sh[55199]: KeyError: '"code"'
Jan 02 21:53:08 raspberrypi sh[55199]: 2023-01-02 21:53:08.65 | ERROR | Application startup failed. Exiting.

Any idea what I did wrong by any chance ?

Thanks a lot !

rootzoll, 2023-01-18:
Can you try an sd card update to the latest RaspiBlitz v1.9.0rc1 which contains the latest LNbits including bugfixes? 
https://github.com/rootzoll/raspiblitz/tree/v1.9#downloading-the-software
MAINMENU > UPDATE > RELEASE

cstenglein, 2023-06-07:
@VeniVidiViciLA  is this still an issue with 1.9.0?

Closing this for now, please re-open if it still persists.