raspiblitz/raspiblitz #3588: lnd/bitcoin 100% sync but node not running

mamar59, 2023-01-03:
Good evening I have launched a rescan of the lnd, since a few days my node no longer wants to pass the synchronization, I tried to meter the new version v1.9 thinking fix my problem but still not, if someone can enlighten me please

thank you in advance

[image]

[image]

mamar59, 2023-01-08:
Hi when I scan the lnd it never ends and I just saw this error what to do?

[ERR] LNWL: Unable to complete chain rescan: -1: Block not found on disk

[image]

rootzoll, 2023-01-13:
Whats the lncli getinfo output? Havent seen that error myself - a search in the lnd github shows: https://github.com/lightningnetwork/lnd/issues/4296#issuecomment-630573181

Solution might be to reindex or delete/resync bitcoin blockchain. Try the `repair` command on RaspiBlitz terminal.

mamar59, 2023-01-14:
Yop! rootzoll thank you for your return, I launched a few days ago a repartion and download of the blockchain, I hope this will solve my problem

[image]

mamar59, 2023-01-16:
Hello after a download of the blockchain and a restart of the node all is finally OK again thank you for your return

closed 2023-01-16 by mamar59