raspiblitz/raspiblitz #3602: mempool not working after password B change

matrix79, 2023-01-10:
changed app password (password B) over normal CLI menu and after rebooting raspiblitz mempool didn't sync new blocks.
checked /home/mempool/mempool/backend/mempool-config.json and noticed, that the old password was still in the config file.

so while lnd.conf and BTC-RPC-Explorer got the right update, mempool config was not proper updated. should get checked and as a bug get fixed if it was not only a special case for me.

flyerbear, 2023-03-03:
This happened to me too (changed password B and mempool stopped syncing). What is the immediate fix until 1.9.0 is released? Should I update the password in `/home/mempool/mempool/backend/mempool-config.json` to my new pw B? Is it just stored in cleartext here?

rootzoll, 2023-03-07:
@flyerbear yes you can edit the password directly in the `mempool-config.json` - ist just cleartext `CORE_RPC.PASSWORD="[PASSWORDB]"`

PR #3619 (2023-01-18): #3602 also change password B for mempool
fix #3602