raspiblitz/raspiblitz #3603: circuitbreaker web ui cannot yet be accessed in raspiblitz

RemoRitzmann, 2023-01-11:
Hi, I have problems setting up circuitbreaker in raspiblitz, it's installed, service is running but I cannot open the web ui. How can I see the circuitbreaker's version? I asked Joost here https://github.com/lightningequipment/circuitbreaker/issues/36 too.

[image]

What is the password for circuitbreaker?

RemoRitzmann, 2023-01-11:
Ok, now I have to figure out how to show web ui.

joostjager, 2023-01-12:
I've added docker CI builds to circuitbreaker and proper version reporting. But not sure if you can run a docker image on raspiblitz?

rootzoll, 2023-01-16:
just merged PR #3599  - will be part of RaspiBlitz v1.9

cstenglein, 2023-06-07:
@RemoRitzmann Have you upgraded to 1.9 and did this fix your issue?

Closing this for now, if it still persists, please re-open

closed 2023-06-07 by cstenglein