raspiblitz/raspiblitz #3604: RaspiBlitz Sync

Lilflowerbed1, 2023-01-11:
I updated the RaspiBlitz img to the latest version. The screen I am seeing has been displaying for days saying "Preparing Blockchain sync please wait..." I asked a telegram group for RaspiBlitz and someone said I'm fine. It's just not showing any percentage of the blockchain sync and wanted to see if anyone can help me out or if this takes days. I've looked around and some people say Raspi is slow so it's taken some weeks to sync. 
[image]

tysecure, 2023-01-12:
Is your SSD in good shape?

Lilflowerbed1, 2023-01-12:
It got delivered from Germany recently. SSD looks fine
I’ll replace when I can to see if that’s the issue, gratitude! I shutdown,
disconnected SSD reconnected and at least now I’m seeing this screen. At
least some of the restarts I want to say are from me tinkering with it.

StellarStoic, 2023-01-13:
I'm seeing the same screen. The first installation I try stuck at blockchain sync at around 40% after 5 days of sync. Then I waited another 2 days and got nothing. An error I already forgot about. (something about a couple of thousands of tries). 

I decided to flash the SD card again and did a fresh install with the existing blockchain on SSD. And I'm here where you are. The web view says Waitsync "Provision done"  

I was hoping for a smoother installation but now I know it's a pain in the butt

Lilflowerbed1, 2023-01-16:
Have you tried joining the telegram group. I wasn't able to get into the english speaking one but got on the spanish speaking RaspiBlitz telegram group.

StellarStoic, 2023-01-16:
I am in the EN telegram group but Telegram with more than 100 members is too much for me. Can't follow the noise. 
here's the link. Try again and wait for shieldy bot to ask you. Confirm that you are human. https://t.me/raspiblitz

closed 2023-06-07 by cstenglein