raspiblitz/raspiblitz #3606: Going nowhere after Sd card flash with ~30% of blockchain sync before the flash.

StellarStoic, 2023-01-13:
What can a man do to continue syncing where it stopped?  I had to flash the SD card because the first sync was stopped at ~30% and was stuck there for 2 days. 
Is there even an option to avoid downloading over TOR which is so painfully slow?

[raspiblitz.log](https://termbin.com/fw5p)

[tail -n 200 /mnt/hdd/bitcoin/debug.log](https://termbin.com/kzgr)

rootzoll, 2023-01-18:
> Is there even an option to avoid downloading over TOR which is so painfully slow?

not yet - there with v1.9.0 there will be an additional I2P download. Switch to clearnet might come in the future.

> What can a man do to continue syncing where it stopped? I had to flash the SD card because the first sync was stopped at ~30% and was stuck there for 2 days.

On sd card update it should offer you the "recovery/update" process which should afer reboot continue the sync where it left off. Even if you do a fresh setup you can choose to use the existing blockchain and continue after final reboot. But if you got stuck on syncing before (and you dont have any funds/channel with the node yet) it might be a cleaner way to  format the SSD clean on a fresh setup - there might have been some data corruption.

StellarStoic, 2023-01-20:
I made it after 2 weeks all good now.
What I did...Downloaded Blockchain with my Pc, transferred it to SSD, and reflashed the raspi.
I documented a few moments on nostr if anyone is interested. npub1glqfhyyt4cy7shjwpse2z2m8f5unpu6gay6v67pp3nd7g5uyhgdqydkp8t
or search for the term raspiblitz.

I'm closing this now

closed 2023-01-20 by StellarStoic