raspiblitz/raspiblitz #3607: cln wallet won't start

robertocop-adc, 2023-01-14:
lightning-cli -H summary:

lightning-cli: Connecting to 'lightning-rpc': Connection refused

RTL is showing:

Error: connect econnrefused 127.0.0.1:6100

Zeus is showing: 

SOCKS server general failure

debug for CLN shows:

*** LAST 30 C-LIGHTNING (MAINNET) INFO LOGS ***
sudo tail -n 30 /home/bitcoin/.lightning/bitcoin/cl.log

2023-01-14T04:20:56.845Z INFO    plugin-backup.py: Killing plugin: exited before replying to getmanifest

2023-01-14T04:21:28.352Z INFO    plugin-backup.py: Killing plugin: exited before replying to getmanifest

2023-01-14T04:21:59.905Z INFO    plugin-backup.py: Killing plugin: exited before replying to getmanifest

2023-01-14T04:22:31.742Z INFO    plugin-backup.py: Killing plugin: exited before replying to getmanifest

2023-01-14T04:23:02.659Z INFO    plugin-backup.py: Killing plugin: exited before replying to getmanifest

2023-01-14T04:23:33.990Z INFO    plugin-backup.py: Killing plugin: exited before replying to getmanifest

C-Lightning options doesn't give the option to close channels.

I'm also still on 1.7.2 and can flash to 1.8.0 if that might help. There is an open channel and I would like to at least recover those funds. Any help would be appreciated.

salenjak, 2023-01-17:
I had a similar issue when c-lightning daemon restarted in the loop.
Also, I didn't manage to connect Zeus, and CRTL was restaring every 30 seconds or something.
The only thing that worked for me was to deactivate Blitz API+WebUI in the node settings.
[image]

rootzoll, 2023-01-18:
> I'm also still on 1.7.2 and can flash to 1.8.0 if that might help. There is an open channel and I would like to at least recover those funds. Any help would be appreciated.

If possible try and sd card update to latest Release Candidate v1.9.0rc1 that you can download here and report if the problem still exists: https://github.com/rootzoll/raspiblitz/tree/v1.9#downloading-the-software

openoms, 2023-01-29:
Yes, you will be ebst to update ,but can try to reinitialize the backup plugin with:
`config,scripts/cl-plugin.baskup.sh on`

openoms, 2023-07-25:
this is dealt with automatically when entering the CLN menu

closed 2023-07-25 by openoms