raspiblitz/raspiblitz #3608: Apps tab won't load on WebUI

HGW39, 2023-01-14:
Upon logging in to the Web UI the Apps tab won't load, just gets stuck in continual loop. Have tried in both Brave and Firefox and same issue occurs. Noticed some other users experiencing this in the Telegram so have raised issue. Please let me know if I can help with any further testing.

[image]

rootzoll, 2023-01-14:
Is this still with the release candidate testversion v1.9.0rc1 or still with v1.8.0?

HGW39, 2023-01-15:
I'm running v1.8.0
[image]

rootzoll, 2023-01-18:
We already fixed a lot on the WebUI/API side .. please try if the sd card update with Release Candidate v1.9.0rc1 already fixes your problem: https://github.com/rootzoll/raspiblitz/tree/v1.9#downloading-the-software

HGW39, 2023-01-23:
Will do and let you know if issue persists.

TheLuvCBD, 2024-03-15:
I'm on version 1.10.0 and I'm having the same problem. It's persisted for a few days now. I've rebooted but it's the same.

fusion44, 2024-03-15:
@TheLuvCBD can you log in to your RaspiBlitz and take a look at the blitz API logs? Looking for any errors in there. Thank you.

sudo journalctl -f blitzapi

bahrthaare, 2024-04-26:
I'm on version 1.11.0 and I'm having the same problem.

bahrthaare, 2024-04-26:
`sudo journalctl -f blitzapi` results in `Failed to add match 'blitzapi': Invalid argument`

rootzoll, 2024-09-27:
There are two other issues reporting similar problems - listing here to be tested/closed if fixed:
- #4431
- #4631

rootzoll, 2024-09-30:
Could be fixed by https://github.com/fusion44/blitz_api/pull/272

rootzoll, 2025-01-04:
reopen for v1.12.0 because a new report says it still happens with latest v1.11.4:
https://github.com/raspiblitz/raspiblitz/issues/4914#issuecomment-2571260464

woiwoiwoiwoi, 2025-01-10:
Not sure if metoos help here, but I'm getting the same. 1.11.4 on Debian VM. Set up that VM specifically to tinker with Alby Hub, but can't even get there.

geco91, 2025-01-31:
@fusion44 please make a catch all exception in that place of the api and coordinate a way with @cstenglein how to give back an error that can be displayed to the user for report.

scarolan, 2025-04-19:
Same issue here, my home and apps tabs both just do this

[image]

rootzoll, 2025-11-11:
Also saw it again on v1.12.0 when there is just Bitcoin. When I install LND the Apps work again.
@fusion44 I tested wit the latest comited API