raspiblitz/raspiblitz #3612: kicked out of raspiblitz telegram, have no idea why?

swidgen, 2023-01-15:
I started using raspiblitz about 3-4 years ago, went on the telegram for some help setting up (mostly hardware issues, needed more compatible stuff) anyway, fast forward to like november of 2022 telegram is different you need a sim card. Ok I join, post my problem then notice there is a pinned message at the top. do what the pinned message says, plus a few other things. I then responded to a number of of other folks having the same problem, pointing out what worked for me, most of it works for them, we say thank you etc… Now I try to ask a question (about 6 week later) and an admin banned me?

Is there another community for general questions that I don’t know about? maybe the telegram was getting too unwieldy?
If there is a good reason like it was decided this needs to be for more technical problems i understand. But as of the moment, even given the least charitable interpretation nothing i said could be seen as rude, or time wasting, trolling etc… it was on topic and I tried to answer people that had the same issues so the serious developers didn’t have to waste their time answering

openoms, 2023-01-15:
Sorry to hear, it must have been a mistake, moderation is a hard job. Share your Telegram username or name here or at t.me/openoms and will get you back.

swidgen, 2023-01-15:
Hey, so at the time my username was just D (could be why i was kicked out).  I changed my name to Doron but i put the username @swidgen if you need the phone number associated with it let me know and I'll DM wherever works best for you

openoms, 2023-01-16:
ok, removed the accidental ban, now you can rejoin at t.me/raspiblitz

swidgen, 2023-01-16:
saw it and joined,  Thanks so much for taking the time, I know you're busy man.