raspiblitz/raspiblitz #3614: " blockchain progress  : 5 restarts "  for 2 days now

billgre, 2023-01-16:
Blockchain Progress : 5 restarts     
 Lightning Progress  : prepare sync       
Please wait - this can take some time 
[image]

Any suggestions?

rootzoll, 2023-01-18:
CTRL+c to exit to terminal and then run `debug` and share the debug report here.

billgre, 2023-01-19:
https://l.termbin.com/quk9wq

billgre, 2023-01-22:
I changed the sd, same result! Now i really need help

billgre, 2023-01-23:
the new v1.9.0rc1 is stuck!!! 
[image]

billgre, 2023-01-25:
I deleted file peers.dat   . and seems that is moving...

billgre, 2023-01-25:
New problem. Asking every 5 min. Password C while syncing. Debug
https://l.termbin.com/wr8i

billgre, 2023-01-28:
Tor, deactivated. Problem solved.  (partially)

rootzoll, 2023-02-03:
TODO: Retest fresh setup process with v1.9 dev

judesen, 2023-02-25:
I am experiencing the exact same issue having flashed SD card with [raspiblitz-v1.8.0c-2022-11-01.img.gz.torrent](https://github.com/rootzoll/raspiblitz/raw/v1.8/home.admin/assets/raspiblitz-v1.8.0c-2022-11-01.img.gz.torrent)

`***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.8.0
chainnetwork: bitcoin / main
 08:10:38 up 14:20,  2 users,  load average: 1.30, 1.73, 1.69

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-02-24 17:51:23 GMT; 14h ago
        CPU: 340ms

Feb 24 17:51:23 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Feb 24 17:51:23 raspberrypi systemd[1]: Failed to start Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Wed 2022-07-20 12:10:49 BST, ends at Sat 2023-02-25 08:10:38 GMT. --
Feb 24 17:51:22 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Feb 24 17:51:23 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Feb 24 17:51:23 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
Feb 24 17:51:23 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 5.
Feb 24 17:51:23 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Feb 24 17:51:23 raspberrypi systemd[1]: bitcoind.service: Start request repeated too quickly.
Feb 24 17:51:23 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Feb 24 17:51:23 raspberrypi systemd[1]: Failed to start Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: inactive (dead)

Feb 24 17:51:19 raspberrypi systemd[1]: Dependency failed for Lightning Network Daemon.
Feb 24 17:51:19 raspberrypi systemd[1]: lnd.service: Job lnd.service/start failed with result 'dependency'.

*** LAST LND (MAINNET) ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Journal begins at Wed 2022-07-20 12:10:49 BST, ends at Sat 2023-02-25 08:10:39 GMT. --
Feb 24 17:51:19 raspberrypi systemd[1]: Dependency failed for Lightning Network Daemon.
Feb 24 17:51:19 raspberrypi systemd[1]: lnd.service: Job lnd.service/start failed with result 'dependency'.

*** LAST 30 LND (MAINNET) INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log

*** CORE LIGHTNING (MAINNET) SYSTEMD STATUS ***
- not activated -

*** BLOCKCHAIN (TESTNET) SYSTEMD STATUS ***
- OFF by config -
*** LND (TESTNET) SYSTEMD STATUS ***
- OFF by config -

*** CORE LIGHTNING (TESTNET) SYSTEMD STATUS ***
- not activated -

*** BLOCKCHAIN (SIGNET) SYSTEMD STATUS ***
[... 351 lines cut ...]
ln_default_running="0"
ln_default_locked="0"
ln_default_version="0.15.4-beta"
btc_default_port=""
btc_default_error_short=""
btc_default_running="0"
btc_default_error_full=""
btc_default_activated="1"
btc_default_sync_initial_done="0"
btc_default_peers="0"
btc_default_ready="0"
btc_default_online="0"
btc_default_version="v23.0.0"
btc_default_sync_percentage="0"

*** LOGFILES ***
Archived and active journals take up 356.0M in the file system.
406M    /var/log

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n`

JDarock, 2023-02-27:
Hi, I'm facing similar issue. I've just yesterday re-flashed again the SD card and from the begining my Raspiblitz was runing without any issues just with the LND installed. But then I started installing other apps as BTCpay Server, Electrum Rust and others and don't remember the exacts moment when after one of the restarts it stopped syncing blockchain and the blockhain progress got stuck and just continuously restarting with no result. Thank you

Here is my debug:
https://l.termbin.com/h4w8

Konrad84, 2023-02-28:
Hi There,

after a awoken night i need help now - i got the same issue .. yesterday afternoon my raspberry showed EXT4-fs errors (device sda1). Blockchain Progress: 6 Restarts .. Reboot doesn´t work. Menu worked so far. I decided to update bitcoin core and nervous i am i forgotten to paste the PGP pubkey .... arr .... then i get Not Started Not Ready Yet No Data in title screen. After that i decided to get another sd flashed with newest version 1.9.0 1rc Fatpack made with Balena Etcher. I tried update Raspiblitz in the Menu two times stuck in Backup process then i turned off the Raspbi Blitz myself and inserted the new image. After Reboot and 6 hours waiting there is still Blockchainsync in Progress .. no idea what to do next please i need help .. 

Greets Thank you.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Konrad84, 2023-02-28:
> 

@billgre -- you fixed it ? Can you tell me what to do im stuck ..

Konrad84, 2023-03-01:
Ok update - this worked for me thanks @openoms ! First i tried to flash SD with 1.9.02rc but that didn´t work for me because get stuck in prepare BC Sync. After discuss in english Raspi group problems with peers.dat and Tor lead to get a new SD card and flashed it with my previous Version 1.8.0c. This worked fine. Now i will close channels and check for hardware issues.

felivix, 2023-03-23:
Hi
I updated to RaspiBlitz v1.9.0rc2 but I'm getting the same CLN error as before with 1.8.0c. 
Now the error reads:
CLN v22.11.1
Not Started | Not Ready Yet | No Data

[image]

The Lightning Progress is stuck and continously displays:
[image]

How can I get CLN to sync again?

BTW, clboss did a great job for several months last year.

Greetings
Felix

rene78, 2023-03-26:
Have the same problem. So I just reflash 1.8.0c and it should work again?

smartbart1, 2023-03-28:
same problem here

judesen, 2023-03-28:
Seems like more and more people are experiencing this issue. I have erased my blockchain file and now waiting for it to download and validate. 

What was interesting was that running with TOR deactivated usually resulted in < 3 peers and constantly losing the connection to peers while enabling TOR gives me > 7 peers while sometimes losing connectivity to the peers.

I’ll give you an update as soon as the blockchain is completed,  it I am currently running at 33.4% after 3 weeks, so it will probably be months.

Godrigos, 2023-03-30:
Just had the same issues. But it's 4 AM. Marking this to read and try fixing it in the morning. Mine says "6 restarts" on Blockchain Progress and some SSD errors.

judesen, 2023-05-01:
I erstattes my raspiblitz node, and noticed that it is shutting down 2 instances of Bitcoind. Is that normal?

I had issues with my node not syncing and I then flashed a new card with the latest version. 

See attached image. Have any of you experienced this before?
[image]

openoms, 2023-05-01:
> I erstattes my raspiblitz node, and noticed that it is shutting down 2 instances of Bitcoind. Is that normal?
> 
> I had issues with my node not syncing and I then flashed a new card with the latest version.
> 

It is not 2 instances, but a simple counter. Normal.

For anyone in this thread to post more info you can press:
CTRL+C and type:
`debug -l`

can post the resulting link here for details.

Finalwar88, 2023-05-08:
same problem since yesterday :/ 

https://l.termbin.com/kz7t

openoms, 2023-05-08:
> same problem since yesterday :/
> 
> https://l.termbin.com/kz7t

```
Error: Invalid or corrupt peers.dat
```
Fix this with running:

```
sudo rm /mnt/hdd/bitcoin/peers.dat
sudo systemctl restart bitcoind
```
Avoid cold restarts. Use a UPS.

openoms, 2023-05-08:
I think showing the bitcoin debug logs and journal could be useful here similar to when failing to unlock the CLN wallet when entering the CLN menu.

Finalwar88, 2023-05-08:
@openoms thanks, total dont saw the error

_"2023-05-08T08:14:53Z Error: Invalid or corrupt peers.dat (Checksum mismatch, data corrupted)."_

seems to work again now.

Godrigos, 2023-11-13:
> > same problem since yesterday :/
> > https://l.termbin.com/kz7t
> 
> ```
> Error: Invalid or corrupt peers.dat
> ```
> 
> Fix this with running:
> 
> ```
> sudo rm /mnt/hdd/bitcoin/peers.dat
> sudo systemctl restart bitcoind
> ```
> 
> Avoid cold restarts. Use a UPS.

Having this issue in 1.10 and the command above can't be execute "Read-only filesystem" error. Rebooting is the only way I'm being capable to restore normal behaviour.

rootzoll, 2023-11-14:
@Godrigos please open a new issue if you still having problems

Godrigos, 2023-11-15:
> @Godrigos please open a new issue if you still having problems

Thanks for the response, but I'll only let a comment here. I know it is a close topic, but will let it here for reference.

I had an idea after the previous post, actually, I remembered having a similar issue in the past and applied the same “solution”. Several months ago I had the same issue with a private file server running Ubuntu Server 20.04 with attached USB3 SSD. Every now and then I had mounting issues with the SSD, the same way I’m having with Raspiblitz, that blocked writing to SSD. After weeks trying to figure out the reason for it (I’m only an informatics enthusiast, not a specialist. I'm a biologist), I noticed that the mounting issues always occurred at peak speed file transfers over internet (above 45MB/s). I capped the max transfer rates at less than 40MB/s and it never happened again with my private server.

I know that USB3 and SSDs should support speeds even greater that that, but it solved the problem for me. Maybe the server network board couldn’t stand such transfer rates. I don’t know, it just works now…

So, two nights ago I did the same with Raspiblitz using my router connection speed administration and since then no mounting issue with the SSD and no more “6 restarts” issues. It’s still early to say it is solved, but I hope it is.

P.s.: The issue always happened at night, probably because the bandwidth was completely available to my server, and now to the Raspiblitz, reaching such higher transfer rates.

Hardware:
Sandisk 2 TB SSD (SDSSDA-2T00-G26 – New). Had the same issue with a 1 TB Sandisk SSD (which is now in my notebook, working fine).
Ugreen USB3 Case (Mod 30848 - Few months of use)
5V 3A Power supply on UPS.