raspiblitz/raspiblitz #3616: Raspiblitz : An error occurred: Unknown error. The response was: 502 .

CrunchySwap, 2023-01-17:
Not quite sure what happened here but node restarted last night for no reason. 
Now I can no longer login via raspiblitz default url https://ip/  and get the following error 

An error occurred: Unknown error. The response was: 502 .

Running bug report shows the following: 

 ```
*** BLITZAPI SYSTEMD STATUS ***
● blitzapi.service - BlitzBackendAPI
     Loaded: loaded (/etc/systemd/system/blitzapi.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2023-01-17 10:54:54 GMT; 52s ago
    Process: 228045 ExecStartPre=/home/admin/config.scripts/blitz.web.api.sh update-config (code=exited, status=0/SUCCESS)
    Process: 228124 ExecStart=/usr/bin/python -m uvicorn app.main:app --port 11111 --host=0.0.0.0 --root-path /api (code=exited, status=1/FAILURE)
   Main PID: 228124 (code=exited, status=1/FAILURE)
        CPU: 400ms

*** LAST BLITZAPI LOGS ***
sudo journalctl -u blitzapi -b --no-pager -n20
-- Journal begins at Tue 2023-01-17 03:02:05 GMT, ends at Tue 2023-01-17 10:55:47 GMT. --
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228098]: sed: couldn't open temporary file ./sed8QQQAR: Permission denied
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228099]: sed: couldn't open temporary file ./sedL2KCcX: Permission denied
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228100]: sed: couldn't open temporary file ./sed67Sq81: Permission denied
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228103]: sed: couldn't open temporary file ./sedhWnANh: Permission denied
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228045]: # CONFIG Web API Lightning --> LND
Jan 17 10:54:53 raspberrypi sudo[228105]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/xxd -ps -u -c 1000 /mnt/hdd/lnd/tls.cert
Jan 17 10:54:53 raspberrypi sudo[228105]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1003)
Jan 17 10:54:53 raspberrypi sudo[228105]: pam_unix(sudo:session): session closed for user root
Jan 17 10:54:53 raspberrypi sudo[228110]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/bitcoin/mainnet/admin.macaroon
Jan 17 10:54:53 raspberrypi sudo[228110]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1003)
Jan 17 10:54:53 raspberrypi sudo[228110]: pam_unix(sudo:session): session closed for user root
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228115]: sed: couldn't open temporary file ./sedA3ttOa: Permission denied
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228116]: sed: couldn't open temporary file ./sed26Jdbe: Permission denied
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228118]: sed: couldn't open temporary file ./sedBlbObo: Permission denied
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228122]: sed: couldn't open temporary file ./sedl1m8QH: Permission denied
Jan 17 10:54:53 raspberrypi blitz.web.api.sh[228045]: # '.env' config updates - blitzapi maybe needs to be restarted
Jan 17 10:54:53 raspberrypi systemd[1]: Started BlitzBackendAPI.
Jan 17 10:54:54 raspberrypi python[228124]: /usr/bin/python: No module named uvicorn
Jan 17 10:54:54 raspberrypi systemd[1]: blitzapi.service: Main process exited, code=exited, status=1/FAILURE
Jan 17 10:54:54 raspberrypi systemd[1]: blitzapi.service: Failed with result 'exit-code'.

 
```

CrunchySwap, 2023-01-17:
https://l.termbin.com/ssboy 

Full debug logs

CrunchySwap, 2023-01-17:
Reinstalling web api gets this error. 

Successfully built psutil
Installing collected packages: wrapt, uvicorn, urllib3, typing-extensions, tenacity, sniffio, setuptools, requests, redis, pyzmq, python-json-logger, pydantic, psutil, protobuf, idna, h11, grpcio-tools, grpcio, googleapis-common-protos, frozenlist, fastapi-plugins, fastapi, deprecated, charset-normalizer, certifi, attrs, aioredis
  Attempting uninstall: wrapt
    Found existing installation: wrapt 1.12.1
ERROR: Cannot uninstall 'wrapt'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.
error='pip install failed'

rootzoll, 2023-01-18:
There was a lot improved on WebUI/API already - can you test with the Release Candidate v1.9.0RC1 if you still have that problem? https://github.com/rootzoll/raspiblitz/tree/v1.9#downloading-the-software

This would need an sd card update - you can later update the same way from the release candidate to the final v1.9.0 version.

cstenglein, 2023-06-07:
@diskawrs is this still present in 1.9.0?

Closing for now, if it still appears, please re-open