raspiblitz/raspiblitz #3622: btcpay feature request: nbxplorer uodate and restart on failure

openoms, 2023-01-19:
As flagged by @nicolasdorier the new btcpay versions will make it easier to install plugins to non-docker systems as well requiring a restart.
Should restart the service on failure.

Also the latest nbxplorer is needed for coming plugin features.

openoms, 2023-01-20:
ok, the btcpayserver.service has:
```
Restart=on-failure
```
already:
https://github.com/rootzoll/raspiblitz/blob/0db034cf4bdc77a533437a72080c013f817754d8/home.admin/config.scripts/bonus.btcpayserver.sh#L576

openoms, 2023-01-21:
A `soft restart` does not exit with failure so need to switch to:
```
Restart = always
```
```
Jan 21 09:03:35 z420 dotnet[1796362]: info: PayServer:      A soft restart has been requested
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      DelayedTransactionBroadcasterHostedService successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      TorServices successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      CheckConfigurationHostedService successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      PullPaymentHostedService successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      NBXplorerListener successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      Disconnected from WebSocket of NBXplorer (BTC)
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      LightningListener successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: BTCPayServer.Payments.Lightning.LightningPendingPayoutListener: LightningPendingPayoutListener successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      Stop watching invoices
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      RatesHostedService successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      DynamicDnsHostedService successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      NewVersionCheckerHostedService successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      DbMigrationsHostedService successfully exited...
Jan 21 09:03:43 z420 systemd[1]: btcpayserver.service: Succeeded.
Jan 21 09:03:43 z420 systemd[1]: btcpayserver.service: Consumed 27.091s CPU time.

₿ sudo systemctl status btcpayserver
● btcpayserver.service - BtcPayServer daemon
     Loaded: loaded (/etc/systemd/system/btcpayserver.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sat 2023-01-21 09:03:43 GMT; 1min 20s ago
    Process: 1795805 ExecStart=/home/btcpay/dotnet/dotnet run --no-launch-profile --no-build -c Release --project >
   Main PID: 1795805 (code=exited, status=0/SUCCESS)
        CPU: 27.091s

Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      Disconnected from WebSocket of NBXplorer (BTC)
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      LightningListener successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: BTCPayServer.Payments.Lightning.LightningPendingPayoutListener: Lightn>
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      Stop watching invoices
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      RatesHostedService successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      DynamicDnsHostedService successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      NewVersionCheckerHostedService successfully exited...
Jan 21 09:03:43 z420 dotnet[1796362]: info: PayServer:      DbMigrationsHostedService successfully exited...
Jan 21 09:03:43 z420 systemd[1]: btcpayserver.service: Succeeded.
Jan 21 09:03:43 z420 systemd[1]: btcpayserver.service: Consumed 27.091s CPU time.
```

openoms, 2023-01-21:
After installing the LNBank plugin makes it fail to a restart loop, this needs more testing:
```
Jan 21 09:29:06 z420 dotnet[2542366]: info: Configuration:  Connection to NBXplorer's database successful, dashboard and reporting features activated.
Jan 21 09:29:06 z420 dotnet[2542366]: info: PayServer:      Starting listening NBXplorer (BTC)
Jan 21 09:29:06 z420 dotnet[2542366]: info: PayServer:      Start watching invoices
Jan 21 09:29:06 z420 dotnet[2542366]: info: PayServer:      Starting payment request expiration watcher
Jan 21 09:29:06 z420 dotnet[2542366]: info: PayServer:      0 pending payment requests being checked since last run
Jan 21 09:29:06 z420 dotnet[2542366]: info: PayServer:      BTC: Checking if any pending invoice got paid while offline...
Jan 21 09:29:06 z420 dotnet[2542366]: Unhandled exception. info: PayServer:      BTC: 0 payments happened while offline
Jan 21 09:29:06 z420 dotnet[2542366]: info: PayServer:      Connected to WebSocket of NBXplorer (BTC)
Jan 21 09:29:06 z420 dotnet[2542366]: System.InvalidOperationException: No backing field could be found for property 'Transaction.Date' and the property does not have a setter.
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Infrastructure.ModelValidator.ValidateFieldMapping(IModel model, IDiagnosticsLogger`1 logger)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Infrastructure.ModelValidator.Validate(IModel model, IDiagnosticsLogger`1 logger)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Infrastructure.RelationalModelValidator.Validate(IModel model, IDiagnosticsLogger`1 logger)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Sqlite.Infrastructure.Internal.SqliteModelValidator.Validate(IModel model, IDiagnosticsLogger`1 logger)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Infrastructure.ModelRuntimeInitializer.Initialize(IModel model, Boolean designTime, IDiagnosticsLogger`1 validationLogger)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Infrastructure.ModelSource.GetModel(DbContext context, ModelCreationDependencies modelCreationDependencies, Boolean designTime)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Internal.DbContextServices.CreateModel(Boolean designTime)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Internal.DbContextServices.get_Model()
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Infrastructure.EntityFrameworkServicesBuilder.<>c.<TryAddCoreServices>b__8_4(IServiceProvider p)
Jan 21 09:29:06 z420 dotnet[2542366]:    at ResolveService(ILEmitResolverBuilderRuntimeContext , ServiceProviderEngineScope )
Jan 21 09:29:06 z420 dotnet[2542366]:    at ResolveService(ILEmitResolverBuilderRuntimeContext , ServiceProviderEngineScope )
Jan 21 09:29:06 z420 dotnet[2542366]:    at ResolveService(ILEmitResolverBuilderRuntimeContext , ServiceProviderEngineScope )
Jan 21 09:29:06 z420 dotnet[2542366]:    at ResolveService(ILEmitResolverBuilderRuntimeContext , ServiceProviderEngineScope )
Jan 21 09:29:06 z420 dotnet[2542366]:    at ResolveService(ILEmitResolverBuilderRuntimeContext , ServiceProviderEngineScope )
Jan 21 09:29:06 z420 dotnet[2542366]:    at ResolveService(ILEmitResolverBuilderRuntimeContext , ServiceProviderEngineScope )
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.Extensions.DependencyInjection.ServiceProvider.GetService(Type serviceType, ServiceProviderEngineScope serviceProviderEngineScope)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.Extensions.DependencyInjection.ServiceLookup.ServiceProviderEngineScope.GetService(Type serviceType)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.Extensions.DependencyInjection.ServiceProviderServiceExtensions.GetRequiredService(IServiceProvider provider, Type serviceType)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.Extensions.DependencyInjection.ServiceProviderServiceExtensions.GetRequiredService[T](IServiceProvider provider)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.DbContext.get_DbContextDependencies()
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.DbContext.get_ContextServices()
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.DbContext.get_InternalServiceProvider()
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.DbContext.Microsoft.EntityFrameworkCore.Infrastructure.IInfrastructure<System.IServiceProvider>.get_Instance()
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.Infrastructure.DatabaseFacade.Microsoft.EntityFrameworkCore.Infrastructure.IInfrastructure<System.IServiceProvider>.get_Instance()
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.RelationalDatabaseFacadeExtensions.GetRelationalService[TService](IInfrastructure`1 databaseFacade)
Jan 21 09:29:06 z420 dotnet[2542366]:    at Microsoft.EntityFrameworkCore.RelationalDatabaseFacadeExtensions.MigrateAsync(DatabaseFacade databaseFacade, CancellationToken cancellationToken)
Jan 21 09:29:06 z420 dotnet[2542366]:    at BTCPayServer.Plugins.LNbank.Services.LNbankPluginMigrationRunner.StartAsync(CancellationToken cancellationToken) in /build/BTCPayServer.Plugins.LNbank/Services/LNbankPluginMigrationRunner.cs:line 37
[... cut ...]
Jan 21 09:29:07 z420 systemd[1]: btcpayserver.service: Consumed 6.264s CPU time.
Jan 21 09:29:07 z420 systemd[1]: Started BtcPayServer daemon.
Jan 21 09:29:08 z420 dotnet[2544775]: info: Configuration:  Data Directory: /home/btcpay/.btcpayserver/Main
Jan 21 09:29:08 z420 dotnet[2544775]: info: Configuration:  Configuration File: /home/btcpay/.btcpayserver/Main/settings.config
Jan 21 09:29:08 z420 dotnet[2544775]: info: BTCPayServer.Plugins.PluginManager: Loading plugins from /home/btcpay/.btcpayserver/Plugins
```

openoms, 2023-01-21:
@rootzoll do you recall  a reason why we are not deleting the whole user folder  (`sudo userdel -rf btcpayserver`) when running `bonus.btcpayserver.sh off` ?

might be for timesaving when testing, but this way a reinstall doesn't get rid of the erroneous plugin.

openoms, 2023-01-21:
With the AOPP plugin the restart works:
```
Jan 21 09:52:32 z420 dotnet[2917073]: info: PayServer:      A soft restart has been requested
Jan 21 09:52:40 z420 dotnet[2917073]: info: PayServer:      DelayedTransactionBroadcasterHostedService successfully exited...
Jan 21 09:52:40 z420 dotnet[2917073]: info: PayServer:      TorServices successfully exited...
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      CheckConfigurationHostedService successfully exited...
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      PullPaymentHostedService successfully exited...
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      NBXplorerListener successfully exited...
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      Disconnected from WebSocket of NBXplorer (BTC)
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      LightningListener successfully exited...
Jan 21 09:52:41 z420 dotnet[2917073]: info: BTCPayServer.Payments.Lightning.LightningPendingPayoutListener: LightningPendingPayoutListener successfully exited...
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      Stop watching invoices
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      RatesHostedService successfully exited...
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      DynamicDnsHostedService successfully exited...
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      NewVersionCheckerHostedService successfully exited...
Jan 21 09:52:41 z420 dotnet[2917073]: info: PayServer:      DbMigrationsHostedService successfully exited...
Jan 21 09:52:41 z420 systemd[1]: btcpayserver.service: Succeeded.
Jan 21 09:52:41 z420 systemd[1]: btcpayserver.service: Consumed 26.007s CPU time.
Jan 21 09:52:41 z420 systemd[1]: btcpayserver.service: Scheduled restart job, restart counter is at 1.
Jan 21 09:52:41 z420 systemd[1]: Stopped BtcPayServer daemon.
Jan 21 09:52:41 z420 systemd[1]: btcpayserver.service: Consumed 26.007s CPU time.
Jan 21 09:52:41 z420 systemd[1]: Started BtcPayServer daemon.
Jan 21 09:52:42 z420 dotnet[3106725]: info: Configuration:  Data Directory: /home/btcpay/.btcpayserver/Main
Jan 21 09:52:42 z420 dotnet[3106725]: info: Configuration:  Configuration File: /home/btcpay/.btcpayserver/Main/settings.config
Jan 21 09:52:42 z420 dotnet[3106725]: info: BTCPayServer.Plugins.PluginManager: Loading plugins from /home/btcpay/.btcpayserver/Plugins
Jan 21 09:52:43 z420 dotnet[3106725]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer - 1.7.4
Jan 21 09:52:43 z420 dotnet[3106725]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.Shopify - 1.7.4
Jan 21 09:52:43 z420 dotnet[3106725]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.Crowdfund - 1.7.4
Jan 21 09:52:43 z420 dotnet[3106725]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.PayButton - 1.7.4
Jan 21 09:52:43 z420 dotnet[3106725]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.PointOfSale - 1.7.4
Jan 21 09:52:43 z420 dotnet[3106725]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer.Plugins.AOPP - 1.0.1
Jan 21 09:52:43 z420 dotnet[3106725]: info: Configuration:  Supported chains: BTC
Jan 21 09:52:43 z420 dotnet[3106725]: info: Configuration:  BTC: Explorer url is http://127.0.0.1:24444/
Jan 21 09:52:43 z420 dotnet[3106725]: info: Configuration:  BTC: Cookie file is /home/btcpay/.nbxplorer/Main/.cookie
Jan 21 09:52:43 z420 dotnet[3106725]: info: Configuration:  Network: Mainnet
Jan 21 09:52:43 z420 dotnet[3106725]: warn: Configuration:  SQLITE backend support is deprecated and will be soon out of support
Jan 21 09:52:45 z420 dotnet[3106725]: info: Configuration:  Root Path: /
Jan 21 09:52:46 z420 dotnet[3106725]: info: PayServer:      Starting listening NBXplorer (BTC)
Jan 21 09:52:46 z420 dotnet[3106725]: info: PayServer:      Start watching invoices
Jan 21 09:52:46 z420 dotnet[3106725]: info: PayServer:      Starting payment request expiration watcher
Jan 21 09:52:46 z420 dotnet[3106725]: info: PayServer:      0 pending payment requests being checked since last run
Jan 21 09:52:46 z420 dotnet[3106725]: info: Configuration:  Now listening on: http://127.0.0.1:23000
Jan 21 09:52:46 z420 dotnet[3106725]: info: PayServer:      BTC: Checking if any pending invoice got paid while offline...
Jan 21 09:52:47 z420 dotnet[3106725]: info: PayServer:      BTC: 0 payments happened while offline
Jan 21 09:52:47 z420 dotnet[3106725]: info: PayServer:      Connected to WebSocket of NBXplorer (BTC)
```

openoms, 2023-01-30:
ok the LNbank install and restart works as expected with postgres  in https://github.com/rootzoll/raspiblitz/pull/3641

rootzoll, 2023-02-03:
waiting for PR #3641 to merge & fix this

closed 2023-05-10 by rootzoll