raspiblitz/raspiblitz #3627: cannot backup SCB to USB from the blitz menu

kittyboy717, 2023-01-22:
when trying to backup the static channel backup to usb drive from the blitz menu this error is received.

[image]

drive is available and formatted to FAT32.

openoms, 2023-01-23:
When there is no extar USB device attached the following command:
```
sudo /home/admin/config.scripts/blitz.backupdevice.sh status
```
Should show:
```
### blitz.backupdevice.sh ###
backupdevice=0
backupCandidates=0
```
Is it the case for you?

kittyboy717, 2023-01-23:
with device removed i receive the following:
```
### blitz.backupdevice.sh ###
backupdevice=0
backupCandidate[0]='zram0 4 GB zram0'
backupCandidates=1
```
even after rebooting, patching, etc.
i'm guess it is the zram issue afterall...

do you still recommend syncing with dev or v1.9??

openoms, 2023-01-23:
yes, it is fixed there. See #3436.

kittyboy717, 2023-01-24:
okay. ^.^
went ahead and flashed an SD with v1.9.
i'll have to wait until i get a clear-ish day before i upgrade (have a bit of added software --  wireguard, charge-lnd, etc.).

i'm sure it'll work since it's already been an issue, but i'll report back afterward to confirm.
ty again for the support~!!

kittyboy717, 2023-01-25:
device added for backup -- as expected.
^.^

closed 2023-01-25 by kittyboy717