raspiblitz/raspiblitz #3631: JAM Orderbook gives 502 error

KainerW, 2023-01-24:
https://i.postimg.cc/KYsmsXwG/orderbook-nginx-error-rb.png

"Error while loading the orderbook. Your current local setup might not support fetching the orderbook. Reason: The server reported a problem: 502 Bad Gateway"

I woke up and had this error in JAM. The day before all worked fine. Nothing has been updated or installed. 

All other apps on the raspiblitz not working flawlessly anymore since then (loading forever, showing half the UI only... e.g. Mempool space shows "Reconnecting" in the upper left corner forever). 

Running 1.9.0RC1 and tried to patch it via "Update - Patch" RB-menue. No help so far.

I guess (not sure) it has to do with enginX:

*** LAST NGINX LOGS ***
sudo journalctl -u nginx -b --no-pager -n20
-- Journal begins at Mon 2023-01-23 06:49:18 GMT, ends at Tue 2023-01-24 11:04:11 GMT. --
Jan 24 10:46:10 raspberrypi systemd[1]: Starting A high performance web server and a reverse proxy server...
Jan 24 10:46:13 raspberrypi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Jan 24 10:46:13 raspberrypi nginx[682]: nginx: [emerg] host not found in upstream "mempool.space" in /etc/nginx/snippets/mempool.conf:39
**Jan 24 10:46:13 raspberrypi nginx[682]: nginx: configuration file /etc/nginx/nginx.conf test failed
Jan 24 10:46:13 raspberrypi systemd[1]: nginx.service: Failed with result 'exit-code'.**
Jan 24 10:46:13 raspberrypi systemd[1]: Failed to start A high performance web server and a reverse proxy server.
Jan 24 10:48:28 raspberrypi systemd[1]: nginx.service: Scheduled restart job, restart counter is at 1.
Jan 24 10:48:28 raspberrypi systemd[1]: Stopped A high performance web server and a reverse proxy server.
Jan 24 10:48:28 raspberrypi systemd[1]: Starting A high performance web server and a reverse proxy server...
Jan 24 10:48:28 raspberrypi systemd[1]: Started A high performance web server and a reverse proxy server.
--> CHECK CONFIG: sudo nginx -t
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful

openoms, 2023-01-27:
make sure to patch to the dev or v1.9 branch in the UPDATE - PATCH menu then reinstall Joinmarket and Jam

KainerW, 2023-01-30:
Thank you Openoms. Tried that, did not really fix/work. 
Made a new raspiblitz-installation on a fresh sd-card. That fixed most the JAM problems. 

NGINX problems from logfile I showed above still exist and I have no idea how or why. Mempool space installation gives "Error loading address data. (Connection to server lost, please retry)" when clicking on an address now. You think that might be related?

openoms, 2023-01-30:
> Thank you Openoms. Tried that, did not really fix/work. Made a new raspiblitz-installation on a fresh sd-card. That fixed most the JAM problems.
> 
> NGINX problems from logfile I showed above still exist and I have no idea how or why.

What version of raspiblitz and Jam are you running now?

> Mempool space installation gives "Error loading address data. (Connection to server lost, please retry)" when clicking on an address now. You think that might be related?

it is simply because you don't have an electrum server installed or synced yet connected to mempool.

cstenglein, 2023-06-09:
@KainerW is that still an issue?

Closing for now, if not please re-open

tioneb22, 2024-06-08:
I'm getting the same error, what was the solution? 

My client is running on StartOS, Bitcoind (version 26.0.0-1) and electrs (version 0.10.4) are fully synced.

Bonzbonanza, 2024-06-18:
Same problem, Umbrel 1.1.2 beta, Bitcoin node 27.0, Electrs 0.10.2.

rootzoll, 2025-02-05:
@openoms can you take a look?

baywind2018, 2025-02-18:
I just installed v0.3.5.1 on a Dell Latitude 7280 and did a collaborative transaction just fine. I get the same "**Error while loading the orderbook. Your current local setup might not support fetching the orderbook. Reason: The server reported a problem: 502 Bad Gateway**" error when trying to go to the order book. Running Bitcoin Core 28.0.0~1 and electrs 0.10.8, both fully sync'd and no errors or notifications. Anyone have any more suggestions?