raspiblitz/raspiblitz #3639: wrong Password - no cl.pw

smamamo, 2023-01-29:
My password C is not working - but works before update core lightning
When I start rasbiblitz  v1.9.0rc1-2022-12-21
connet via ssh 
Enter password 
[image]
I enter the correct password and get "wrong password"
[image]
When I go in menu Password 
[image]
[image]
I get "error='no password c1'"
[image]

Software log
[9g52.pdf](https://github.com/rootzoll/raspiblitz/files/10530291/9g52.pdf)

sudo tail -n 30 -f /home/bitcoin/.lightning/bitcoin/cl.log
[image]

any idea what I make wrong

many thanks for you help

openoms, 2023-01-29:
Couldn't reproduce this yet.

Try to enter the REPAIR-CL menu and use DECRYPT to remove the password then can set  a new one with ENCRYPT.
to access the option type:
`menu`
CLN -> REPAIR-CL

smamamo, 2023-01-29:
When I go - menu / CLN = "Enter Password" - wrong Password
when I try - menu / Repair CL/DECRYPT  =   "Enter Password" - wrong Password
shred: /home/bitcoin/.cl.pw: failed to open for writing: No such file or directory
# Cancelled
# get raspiblitz config
# Running 'cl.hsmtool.sh decrypt mainnet'
Enter hsm_secret password:
Enter hsm_secret password:
lightning-hsmtool: Could not retrieve the seed. Wrong password ?
# Running 'cl.hsmtool.sh unlock mainnet'
# Wrong password
shred: /home/bitcoin/.cl.pw: failed to open for writing: No such file or directory
# Cancelled
Enter hsm_secret password:
cat: /home/bitcoin/.cl.pw: No such file or directory
lightning-hsmtool: Could not read pass from stdin.
# Couldn't decrypt
SIGINT or SIGTERM received, exiting...

config.scripts/cl-plugin.backup.sh on
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Updating 4d3560b..1b8d5dd
Fast-forward
 .ci/test.py                                     |   70 +-
 .github/workflows/main.yml                      |   39 +-
 .gitmodules                                     |    3 +
 README.md                                       |   15 +-
 {commando => archive/commando}/README.md        |    0
 {commando => archive/commando}/commando.py      |    0
 {commando => archive/commando}/requirements.txt |    0
 {commando => archive/commando}/test_commando.py |    0
 autopilot/autopilot.py                          |   25 +-
 autopilot/c-lightning-autopilot.py              |    4 +-
 autopilot/lib_autopilot.py                      |   17 +-
 autopilot/poetry.lock                           | 1091 -----------------
 autopilot/pyproject.toml                        |   23 +-
 autopilot/test_autopilot.py                     |    4 +-
 backup/backend.py                               |    2 +-
 backup/backends.py                              |    6 +-
 backup/backup.py                                |    8 +-
 backup/filebackend.py                           |    8 +-
 backup/poetry.lock                              | 1416 ++++++++++++-----------
 backup/protocol.py                              |    9 +-
 backup/pyproject.toml                           |   14 +-
 backup/requirements.txt                         |    2 -
 backup/server.py                                |   19 +-
 backup/socketbackend.py                         |   48 +-
 backup/test_backup.py                           |   75 +-
 donations/poetry.lock                           | 1088 +++++++++++++++++
 donations/pyproject.toml                        |   23 +
 donations/requirements-dev.txt                  |    1 -
 donations/requirements.txt                      |    7 -
 drain/README.md                                 |   10 +-
 drain/clnutils.py                               |   23 +
 drain/drain.py                                  |   15 +-
 drain/requirements.txt                          |    1 -
 drain/test_drain.py                             |    1 -
 feeadjuster/README.md                           |    2 +
 feeadjuster/clnutils.py                         |   23 +
[... 44 lines cut ...]
 create mode 100644 paytest/poetry.lock
 create mode 100644 rebalance/clnutils.py
 delete mode 100644 request-invoice/README.md
 delete mode 100755 request-invoice/requestinvoice.py
 delete mode 100644 request-invoice/requirements.txt
 delete mode 100644 request-invoice/test_request-invoice.py
HEAD is now at 4d3560b reqirements: switch to pyln-client 0.12 for backup, drain, feeadj, rebalance and summary
# Checking dependencies
# Stop the lightningd.service
ls: cannot access '/home/bitcoin/.lightning/bitcoin/backup.lock': No such file or directory
# Initialize the backup plugin
Traceback (most recent call last):
  File "/home/bitcoin/cl-plugins-available/plugins/backup/backup-cli", line 2, in <module>
    from backends import get_backend
  File "/home/bitcoin/cl-plugins-available/plugins/backup/backends.py", line 5, in <module>
    from backend import Backend
  File "/home/bitcoin/cl-plugins-available/plugins/backup/backend.py", line 7, in <module>
    from tqdm import tqdm
ModuleNotFoundError: No module named 'tqdm'
# The crontab for admin now is:
@weekly /home/admin/config.scripts/cl-plugin.backup.sh backup-compact mainnet
# Started the lightningd.service

deactivate CLBOSS 
I all ready try - no result  - still ask for en

smamamo, 2023-01-30:
When I try to DECRYPT comes

“Enter Password“ / wrong Password

openoms, 2023-01-30:
If it is still a problem after a full restart try to double check you password by typing to Notepad and paste it to the terminal.

smamamo, 2023-01-30:
I alway past it in the terminal, so I try the other way and typing to terminal. And yes i typing to notepad (textEdit) and paste it . = wrong password

What I don’t understand is when I go to menu /  PASSWORD / CL / I get 

# FAIL: there is no password ‚c'
error='no password c1' 
	
> Am 30.01.2023 um 08:05 schrieb openoms ***@***.***>:
> 
> If it is still a problem after a full restart try to double check you password by typing to Notepad and paste it to the terminal.
> 
>

openoms, 2023-01-30:
You can try to decrypt manually with the command:
```
sudo -u bitcoin lightning-hsmtool decrypt /home/bitcoin/.lightning/bitcoin//hsm_secret
```

smamamo, 2023-01-31:
sudo -u bitcoin lightning-hsmtool decrypt /home/bitcoin/.lightning/bitcoin//hsm_secret
Enter hsm_secret password:
lightning-hsmtool: Could not retrieve the seed. Wrong password ?
[image]

openoms, 2023-01-31:
This suggests that the password given is wrong. 

Do you have any channels?

If there are no channels you can just restore the wallet (hsm_secret) from the seed.
If there are channels you should not reset the wallet to keep the channel database.

You could then restore the `hsm_secret` from seed in a separate location and copy the file in place.

If you are not familiar with linux can give more guidance here.

smamamo, 2023-01-31:
Yes I had channels open and asking for instructions on how to recover.

openoms, 2023-01-31:
Referring to this: https://github.com/rootzoll/raspiblitz/blob/v1.8/FAQ.cl.md#how-to-test-the-seedwords

The lines starting with # are comments about what the commands do. 
This only touches the onchain part of your CLN wallet, the hsm_secret.
Please follow accurately:
```
# stop CLN
sudo systemctl stop lightningd

# change to the bitcoin user
sudo su - bitcoin

# generate the hsm_secret in temporary directory from your CLN seed words (follow the instructions)
lightning-hsmtool generatehsm /dev/shm/hsm_secret

# backup your old hsm_secret
mv /home/bitcoin/.lightning/bitcoin/hsm_secret /home/bitcoin/.lightning/bitcoin/hsm_secret_old

# move the new hsm_secret in place
mv /dev/shm/hsm_secret /home/bitcoin/.lightning/bitcoin/

# back to admin
exit

# start lightningd
sudo systemctl start ligthningd

# show the logs
cllog
````

smamamo, 2023-01-31:
Hmm  
admin@192.168.254.61:~ ₿ sudo systemctl stop lightningd
admin@192.168.254.61:~ ₿ sudo su - bitcoin
-bash: /home/bitcoin/.cargo/env: No such file or directory
-bash: /home/bitcoin/.cargo/env: No such file or directory

openoms, 2023-01-31:
no you didn't, can ignore that.
type:
`clear`
to clear the screen

smamamo, 2023-01-31:
something go wrong here clog

bitcoin@raspberrypi:~ $ mv /home/bitcoin/.lightning/bitcoin/hsm_secret /home/bitcoin/.lightning/bitcoin/hsm_secret_old
bitcoin@raspberrypi:~ $ mv /dev/shm/hsm_secret /home/bitcoin/.lightning/bitcoin/
mv: replace '/home/bitcoin/.lightning/bitcoin/hsm_secret', overriding mode 0400 (r--------)? 
bitcoin@raspberrypi:~ $ exit
logout
admin@192.168.254.61:~ ₿ sudo systemctl start ligthningd
Failed to start ligthningd.service: Unit ligthningd.service not found.
admin@192.168.254.61:~ ₿ cllog

2023-01-31T21:54:31.494Z DEBUG   lightningd: Opened log file cl.log
2023-01-31T21:54:31.506Z DEBUG   plugin-manager: started(136645) /usr/local/libexec/c-lightning/plugins/autoclean
2023-01-31T21:54:31.518Z DEBUG   plugin-manager: started(136647) /usr/local/libexec/c-lightning/plugins/chanbackup
2023-01-31T21:54:31.531Z DEBUG   plugin-manager: started(136650) /usr/local/libexec/c-lightning/plugins/bcli
2023-01-31T21:54:31.542Z DEBUG   plugin-manager: started(136652) /usr/local/libexec/c-lightning/plugins/commando
2023-01-31T21:54:31.553Z DEBUG   plugin-manager: started(136654) /usr/local/libexec/c-lightning/plugins/fetchinvoice
2023-01-31T21:54:31.571Z DEBUG   plugin-manager: started(136655) /usr/local/libexec/c-lightning/plugins/funder
2023-01-31T21:54:31.583Z DEBUG   plugin-manager: started(136657) /usr/local/libexec/c-lightning/plugins/topology
2023-01-31T21:54:31.594Z DEBUG   plugin-manager: started(136659) /usr/local/libexec/c-lightning/plugins/keysend
2023-01-31T21:54:31.605Z DEBUG   plugin-manager: started(136660) /usr/local/libexec/c-lightning/plugins/offers
2023-01-31T21:54:31.621Z DEBUG   plugin-manager: started(136663) /usr/local/libexec/c-lightning/plugins/pay
2023-01-31T21:54:31.632Z DEBUG   plugin-manager: started(136666) /usr/local/libexec/c-lightning/plugins/txprepare
2023-01-31T21:54:31.644Z DEBUG   plugin-manager: started(136667) /usr/local/libexec/c-lightning/plugins/spenderp
2023-01-31T21:54:31.657Z DEBUG   plugin-manager: started(136669) /usr/local/libexec/c-lightning/plugins/bookkeeper
2023-01-31T21:54:31.669Z DEBUG   plugin-manager: started(136671) /home/bitcoin/cl-plugins-available/plugins/backup/backup.py
2023-01-31T21:54:32.454Z INFO    plugin-backup.py: Killing plugin: exited before replying to getmanifest
2023-01-31T21:54:32.454Z DEBUG   lightningd: io_break: check_plugins_manifests
2023-01-31T21:54:32.454Z DEBUG   lightningd: io_loop_with_timers: plugins_init
2023-01-31T21:54:32.455Z DEBUG   lightningd: PROMPT: The hsm_secret is encrypted with a password. In order to decrypt it and start the node you must provide the password.
2023-01-31T21:54:32.455Z DEBUG   lightningd: PROMPT: Enter hsm_secret password:
2023-01-31T21:54:32.455Z DEBUG   lightningd: PROMPT: 
2023-01-31T21:54:35.224Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_channeld
2023-01-31T21:54:35.228Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_closingd
2023-01-31T21:54:35.232Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_connectd
2023-01-31T21:54:35.236Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_gossipd
2023-01-31T21:54:35.239Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_hsmd
2023-01-31T21:54:35.243Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_onchaind
2023-01-31T21:54:35.246Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_openingd
2023-01-31T21:54:35.250Z DEBUG   hsmd: pid 137041, msgfd 44

smamamo, 2023-01-31:
after clear - I start again with 
lightning-hsmtool generatehsm /dev/shm/hsm_secret
?

smamamo, 2023-01-31:
bitcoin@raspberrypi:~ $ mv -f /dev/shm/hsm_secret /home/bitcoin/.lightning/bitcoin/
bitcoin@raspberrypi:~ $ exit
logout
admin@192.168.254.61:~ ₿ sudo systemctl start ligthningd
Failed to start ligthningd.service: Unit ligthningd.service not found.
admin@192.168.254.61:~ ₿ cllog
2023-01-31T22:56:14.736Z DEBUG   plugin-manager: started(203459) /usr/local/libexec/c-lightning/plugins/txprepare
2023-01-31T22:56:14.746Z DEBUG   plugin-manager: started(203461) /usr/local/libexec/c-lightning/plugins/spenderp
2023-01-31T22:56:14.758Z DEBUG   plugin-manager: started(203462) /usr/local/libexec/c-lightning/plugins/bookkeeper
2023-01-31T22:56:14.767Z DEBUG   plugin-manager: started(203468) /home/bitcoin/cl-plugins-available/plugins/backup/backup.py
2023-01-31T22:56:15.730Z INFO    plugin-backup.py: Killing plugin: exited before replying to getmanifest
2023-01-31T22:56:15.730Z DEBUG   lightningd: io_break: check_plugins_manifests
2023-01-31T22:56:15.731Z DEBUG   lightningd: io_loop_with_timers: plugins_init
2023-01-31T22:56:15.731Z DEBUG   lightningd: PROMPT: The hsm_secret is encrypted with a password. In order to decrypt it and start the node you must provide the password.
2023-01-31T22:56:15.731Z DEBUG   lightningd: PROMPT: Enter hsm_secret password:
2023-01-31T22:56:15.731Z DEBUG   lightningd: PROMPT: 
2023-01-31T22:56:18.687Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_channeld
2023-01-31T22:56:18.690Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_closingd
2023-01-31T22:56:18.696Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_connectd
2023-01-31T22:56:18.701Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_gossipd
2023-01-31T22:56:18.705Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_hsmd
2023-01-31T22:56:18.708Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_onchaind
2023-01-31T22:56:18.714Z DEBUG   lightningd: testing /usr/local/libexec/c-lightning/lightning_openingd
2023-01-31T22:56:18.720Z DEBUG   hsmd: pid 203921, msgfd 44
2023-01-31T22:56:22.428Z DEBUG   connectd: pid 204085, msgfd 48
2023-01-31T22:56:22.429Z DEBUG   hsmd: Client: Received message 27 from client
2023-01-31T22:56:22.429Z DEBUG   hsmd: new_client: 0
2023-01-31T22:56:22.431Z DEBUG   0205ca48173a75f83aee8c6edac9a09fedd7bf9f85e4e48a059ed54261693525e8-hsmd: Got WIRE_HSMD_ECDH_REQ
2023-01-31T22:56:22.431Z DEBUG   hsmd: Client: Received message 1 from client
2023-01-31T22:56:22.519Z DEBUG   connectd: Proxy address: 127.0.0.1:9050
2023-01-31T22:56:22.519Z DEBUG   connectd: Created listener on 127.0.0.1:9736
2023-01-31T22:56:22.519Z INFO    connectd: Static Tor service onion address: "yawkkhdyk4ptnbaeaga3fcbe52makqjmlslxkvsrfhlssbcgyu4isfid.onion:9736,127.0.0.1:9736" bound from extern port 9736 
2023-01-31T22:56:22.520Z DEBUG   connectd: REPLY WIRE_CONNECTD_INIT_REPLY with 0 fds
2023-01-31T22:56:22.520Z DEBUG   lightningd: io_break: connect_init_done
2023-01-31T22:56:22.520Z DEBUG   lightningd: io_loop: connectd_init
2023-01-31T22:56:22.520Z **BROKEN** wallet: Wallet node_id does not match HSM: 03e207d06c774ecc8009a93623c85d33cfee8bcc570d1e8a078a724b0d59b70afc != 0205ca48173a75f83aee8c6edac9a09fedd7bf9f85e4e48a059ed54261693525e8. Did your hsm_secret change?

openoms, 2023-02-01:
there was a typo:
use:
```
sudo systemctl start lightningd
```
or can do restart now:
```
sudo systemctl restart lightningd
```
then check the logs again with:
```
cllog
```

smamamo, 2023-02-07:
after reboot from a  new sd card with raspiblitz-v1.8.0c-2022-11-01 and cln - v0.11.2
no more Channel no more funds but on 1ml.com are channels and funds 
Public Key: 03e207d06c774ecc8009a93623c85d33cfee8bcc570d1e8a078a724b0d59b70afc
any link to help for recovering?

openoms, 2023-02-07:
There are notes here:
https://github.com/rootzoll/raspiblitz/blob/dev/FAQ.cl.md#recovery

if you have the correct 24 words (CLN seed) you can recover the onchain funds.

The channel state is replicated to the SDcard so your old SDcard has a backup. Keep it if you have it.

Also if you have successfully started a recent CLN version you will have an emergency.recover file which can ping your peers to force close even if you have lost the channel states. It also needs the 24 words to recover the hsm_secret first. 
Links given here: https://github.com/rootzoll/raspiblitz/issues/3303#issuecomment-1228295389

https://blog.blockstream.com/core-lightning-v0-12-0/
Video demo: https://www.youtube.com/watch?v=_oaJc4ISzTE

there is an emergency.recover file in /home/bitcoin/.lightning/bitcoin near the hsm_secret
this is on the disk
There should be a copy on the old SDcard also.

If you are stuck and would want to avoid sharing in public here can DM me on 
keybase.io/oms (encrypted) or t.me/openoms (unencrypted). Beware of impersonators when writing in private.

openoms, 2023-02-07:
I see that your channels are all pretty much force closed (might be out of date info on explorers) https://amboss.space/node/03e207d06c774ecc8009a93623c85d33cfee8bcc570d1e8a078a724b0d59b70afc

so you have a good chance to be able to restore funds with just the seed.

openoms, 2023-02-10:
The funds have been recovered manually.
The resulting documentation is in #3666.

closed 2023-02-10 by openoms