raspiblitz/raspiblitz #3640: Mempool and Thunderhub can not be installed

StellarStoic, 2023-01-29:
My issue is that every installation try of a Mempool and Tunderhub has failed on **V1.8 patched** 
The 32GB SD card is new, fast, and durable. I'm running Blitz over wifi on Raspberry 4b 8Gb, 2TB SSD

As I said. installing Mempool or Thunderhub always failed.
[image]
[image]

Here are the logs...

`sudo cat /var/cache/raspiblitz/debug.log | nc termbin.com 9999`
raspiblitz debug logs: https://termbin.com/3oefe 
Mempool installation log: https://termbin.com/zkux
Thunderhub installation log: https://termbin.com/ricl7 (Line 3282 gives an error about connection but I am sure the connection is just fine and I'm not behind any proxy)

Then I upgraded the Raspiblitz to **v1.9.0rc1** as Openoms suggested
Raspiblitz debug logs: https://termbin.com/tlqu

I could install Thunderhub (The status on webUI says installation successful and even got the links to the service) but couldn't run it (I reinstalled it two times and could not get it to run. Log was always the same "This site can’t be reached")
thunderbird log: https://termbin.com/atio

When trying to install Mempool, it said it is already installed but it wasn't (No logs, just a picture)

[image]

Openoms also suggested updating node version with `config.scripts/bonus.nodejs.sh on`
Old node version: v16.14.2
node version after update: 18.12.0
I also did `config.scripts/bonus.nodejs.sh off` and "on" again

There were no changes doing that. I still can't install mempool and thundehub .

For example what Thunderhug log is complaining about after all that...

`*** LAST 20 Thunderhub LOGS ***
sudo journalctl -u thunderhub -b --no-pager -n20
-- Journal begins at Sun 2023-01-29 04:34:17 GMT, ends at Sun 2023-01-29 16:03:2        6 GMT. --
Jan 29 16:02:45 raspberrypi systemd[1]: thunderhub.service: Scheduled restart jo        b, restart counter is at 4.
Jan 29 16:02:45 raspberrypi systemd[1]: Stopped ThunderHub daemon.
Jan 29 16:02:45 raspberrypi systemd[1]: thunderhub.service: Consumed 1.603s CPU         time.
Jan 29 16:02:45 raspberrypi systemd[1]: Started ThunderHub daemon.
Jan 29 16:02:46 raspberrypi npm[679854]: > thunderhub@0.13.6 start
Jan 29 16:02:46 raspberrypi npm[679854]: > cross-env NODE_ENV=production nest start
Jan 29 16:02:46 raspberrypi npm[679978]: sh: 1: cross-env: not found
Jan 29 16:02:46 raspberrypi systemd[1]: thunderhub.service: Main process exited,code=exited, status=127/n/a
Jan 29 16:02:46 raspberrypi systemd[1]: thunderhub.service: Failed with result 'exit-code'.
Jan 29 16:02:46 raspberrypi systemd[1]: thunderhub.service: Consumed 1.690s CPUtime.
Jan 29 16:03:17 raspberrypi systemd[1]: thunderhub.service: Scheduled restart job, restart counter is at 5.
Jan 29 16:03:17 raspberrypi systemd[1]: Stopped ThunderHub daemon.
Jan 29 16:03:17 raspberrypi systemd[1]: thunderhub.service: Consumed 1.690s CPU time.
Jan 29 16:03:17 raspberrypi systemd[1]: Started ThunderHub daemon.
Jan 29 16:03:18 raspberrypi npm[684376]: > thunderhub@0.13.6 start
Jan 29 16:03:18 raspberrypi npm[684376]: > cross-env NODE_ENV=production nest start
Jan 29 16:03:19 raspberrypi npm[684728]: sh: 1: cross-env: not found
Jan 29 16:03:19 raspberrypi systemd[1]: thunderhub.service: Main process exited,code=exited, status=127/n/a
Jan 29 16:03:19 raspberrypi systemd[1]: thunderhub.service: Failed with result 'exit-code'.
Jan 29 16:03:19 raspberrypi systemd[1]: thunderhub.service: Consumed 1.896s CPU time.
`

Next, I will start fresh and update node to v18.12.9 before installing other apps.

To be continued...

openoms, 2023-01-29:
A few things here:
the first screenshot suggests a connection problem - maybe related to VPN / ISP?

when there are permission errors often using `sudo` is the solution, but thsi shouldn't be necessary for the pinned versions. 

The install works on my RPi on the dev branch so couldn't reproduce this so far.

Consider trying again a bit later and see if anyone else has this issue.

StellarStoic, 2023-01-30:
Mempool has been successfully installed after deleting `/home/mempool/mempool/frontend/node_modules` folder.

StellarStoic, 2023-02-04:
I repeated the process a few time. Fresh instalation of v1.8+patch (No way I can install mempool and thunderhub on this stable versions). 
On v1.9rc1 mempool installed smothly (Could be because of the newer node version 18.12.9) but I still can't install thunderhub I always get some weird errors https://termbin.com/8b3z.

StellarStoic, 2023-02-04:
I will close this issue since I think that's unique to my setup.
RTL is working so Thunderhub is not really necesery for me atm. 
I will post if I get some new clues.

closed 2023-02-04 by StellarStoic

StellarStoic, 2023-02-12:
After idk how many tries, I finally installed everything that Raspiblitz app stack has to offer including the ones I have issues with. 

What makes this happen...

Flashed 1.8.0 version > +patch > upgrade to 1.9.0rc2

I upgraded npm with `sudo npm install -g npm@9.4.2`

Then I followed this [tutorial](https://thisdavej.com/upgrading-to-more-recent-versions-of-node-js-on-the-raspberry-pi/)  where I upgraded the nodejs to the recent version which is now _v18.12.0_

I did run the `sudo apt update` command along the process and it could be that the issue was there the whole time because the Pi's dependencies weren't updated. I never run this command before.

dskvr, 2023-02-19:
I am having a similar problem. When thunderhub is installed via raspiblitz GUI, it reports successful installation but port `3010` is not accessible.

StellarStoic, 2023-02-20:
> I am having a similar problem. When thunderhub is installed via raspiblitz GUI, it reports successful installation but port `3010` is not accessible. 

I didn't have issues with ports. Check firewall or look if something else is occupied this port number.