raspiblitz/raspiblitz #3646: LND stuck at "Scanning Blocks: 100.00" + Electrum: Not running...

henkisdabro, 2023-01-31:
I have been seeing this screen for a long time now and want to try to resolve it. Not sure of the reasons. Happy to provide more details if needed! Thanks in advance for any tips!

`debug -l` output: https://l.termbin.com/5105f

`sudo journalctl -u electrs` outputs:

```
-- Journal begins at Mon 2023-01-30 17:56:06 GMT, ends at Tue 2023-01-31 06:31:05 GMT. --
Jan 30 17:56:22 raspberrypi electrs[1098668]: Starting electrs 0.9.7 on aarch64 linux with Config { network: Bitcoin, db_path: "/mnt/hdd/app-storage/electrs/db/bitcoin", daemon_dir: "/home/electrs/.bitcoin", daemon_auth: UserPass("raspibolt", "<sensitive>"), d>
Jan 30 17:56:22 raspberrypi electrs[1098668]: [2023-01-30T17:56:22.382Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
Jan 30 17:56:22 raspberrypi electrs[1098668]: [2023-01-30T17:56:22.382Z INFO  electrs::server] serving Electrum RPC on 0.0.0.0:50001
Jan 30 17:56:22 raspberrypi electrs[1098668]: [2023-01-30T17:56:22.436Z INFO  electrs::db] "/mnt/hdd/app-storage/electrs/db/bitcoin": 153 SST files, 37.599674429 GB, 4.69618748 Grows
Jan 30 17:56:27 raspberrypi electrs[1098668]: [2023-01-30T17:56:27.308Z INFO  electrs::chain] loading 761248 headers, tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f
Jan 30 17:56:32 raspberrypi electrs[1098668]: [2023-01-30T17:56:32.041Z INFO  electrs::chain] chain updated: tip=0000000000000000000640e114b9e6272b9688178273e193ef957e02fad1069f, height=761248
Jan 30 17:56:32 raspberrypi electrs[1098668]: [2023-01-30T17:56:32.113Z INFO  electrs::index] indexing 2000 blocks: [761249..763248]
Jan 30 17:56:32 raspberrypi electrs[1098668]: thread 'p2p_loop' panicked at 'invalid message: OversizedVectorAllocation { requested: 12000072, max: 4000000 }', src/p2p.rs:250:70
Jan 30 17:56:32 raspberrypi electrs[1098668]: note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Jan 30 17:56:32 raspberrypi electrs[1098668]: [2023-01-30T17:56:32.150Z INFO  electrs::db] closing DB at /mnt/hdd/app-storage/electrs/db/bitcoin
Jan 30 17:56:32 raspberrypi electrs[1098668]: Error: electrs failed
```

Is it possibly related to this issue?
https://github.com/romanz/electrs/issues/783

Screenshots from my Raspi:
[image]
[image]

openoms, 2023-02-03:
You need to update lnd and electrs.
Can use the latest v1.8.0c image or the UPDATE menu for LND.

For electrs need to run:
`patch`
and reinstall.

closed 2023-02-03 by openoms

henkisdabro, 2023-02-04:
Thank you, a reflash of the card using the 1.8.0c image indeed did the trick. I could not resolve it by updating LND and running the Patch feature, the problem seemed to persist, but I Probably did something wrong. Appreciate the advice, it's all green now on my display, cheers!