raspiblitz/raspiblitz #3674: 1.9.0rc2 CLN grpc plugin not installed

cstenglein, 2023-02-12:
Looks like the cln grpc plugin is not installed with 1.9.0rc2

fusion44, 2023-02-12:
For some reason on my node gRPC plugin is activated already:
[image]

Normally it should be enough to add `grpc-port=4772` to the core lightning config.

openoms, 2023-02-12:
It is activated and preinstalled in the fatpack:
```
lightning-cli plugin list

...
      {
         "name": "/home/bitcoin/cl-plugins-available/cln-grpc",
         "active": true,
         "dynamic": false
      },
...
```

During provision:
https://github.com/rootzoll/raspiblitz/blob/b91506b4fb7e6d41edb8c35c4803369af920d671/home.admin/_provision_.sh#L320

recovery:
https://github.com/rootzoll/raspiblitz/blob/b91506b4fb7e6d41edb8c35c4803369af920d671/home.admin/_provision.setup.sh#L347

SETTINGS:
https://github.com/rootzoll/raspiblitz/blob/b91506b4fb7e6d41edb8c35c4803369af920d671/home.admin/00settingsMenuBasics.sh#L445

closed 2023-05-10 by rootzoll