raspiblitz/raspiblitz #3675: 1.9.0rc2 error on initializing backup with CLN installation

cstenglein, 2023-02-13:
Switch on CLN in the menu:
```
# Checking dependencies
# Stop the lightningd.service
ls: cannot access '/home/bitcoin/.lightning/bitcoin/backup.lock': No such file or directory
# Initialize the backup plugin
Traceback (most recent call last):
  File "/home/bitcoin/cl-plugins-available/plugins/backup/backup-cli", line 2, in <module>
    from backends import get_backend
  File "/home/bitcoin/cl-plugins-available/plugins/backup/backends.py", line 5, in <module>
    from backend import Backend
  File "/home/bitcoin/cl-plugins-available/plugins/backup/backend.py", line 7, in <module>
    from tqdm import tqdm
ModuleNotFoundError: No module named 'tqdm'
# The crontab for admin now is:
@weekly /home/admin/config.scripts/cl-plugin.backup.sh backup-compact mainnet
```

openoms, 2023-02-13:
here is a dependency missing for the backup plugin , will try to reproduce.

Surenic, 2023-02-14:
Same issue here, all of a sudden on lightningd restart (1.9.0rc1/2)

openoms, 2023-02-19:
This should be fixed with a reflash containing #3684, but to just document the dirty metod I managed to fix the error with my running v1.8.0 node like this:

```
# patch to the branch of rootzoll `backup-plugin-update` (or dev once #3684 is merged)
# uninstall old poetry
sudo -u bitcoin pip uninstall poetry
# run the backup plugin install
config.scripts/cl-plugin.backup.sh on
# see if CLN is running ok
cllog
sudo journalctl -fu ligthningd
```

if still errorred with something like:
```
    file:///home/bitcoin/${netprefix}lightningd.sqlite3.backup                                                     
Traceback (most recent call last):                                                                                 
  File "/usr/local/bin/poetry", line 5, in <module>                                                                
    from poetry.console.application import main
  File "/usr/local/lib/python3.9/dist-packages/poetry/console/application.py", line 12, in <module>
    from cleo.events.console_command_event import ConsoleCommandEvent
ModuleNotFoundError: No module named 'cleo.events'
```

reinit manually on mainnet:
```
CLNETWORK=bitcoin
netprefix=""
sudo -u bitcoin poetry run ./backup-cli init --lightning-dir /home/bitcoin/.lightning/${CLNETWORK}     file:///home/bitcoin/${netprefix}lightningd.sqlite3.backup
```

Expected:
```
Found an existing database at /home/bitcoin/.lightning/bitcoin/lightningd.sqlite3, initializing the backup with a snapshot
Successfully written initial snapshot to file:///home/bitcoin/lightningd.sqlite3.backup
Initialized backup backend file:///home/bitcoin/lightningd.sqlite3.backup, you can now start Core-Lightning
```

rootzoll, 2023-03-07:
Waiting for PR #3684 to be merged

closed 2023-05-19 by rootzoll