raspiblitz/raspiblitz #3676: 1.9.0rc2 cln installation stuck

cstenglein, 2023-02-13:
To reproduce: Switch on cln in the menu. Installation doesn't move forward from the following point:

```
# The installed Core Lightning version is: v22.11.1

# To activate the aliases reopen the terminal or use:
source ~/_aliases
# Monitor the lightningd with:
sudo journalctl -fu lightningd
sudo systemctl status lightningd
# logs:
sudo tail -f /home/bitcoin/.lightning/bitcoin/cl.log
# for the command line options use
lightning-cli help
```

openoms, 2023-02-13:
The next steps here are:
```
/home/admin/config.scripts/cl-plugin.cln-grpc.sh install
/home/admin/config.scripts/cl-plugin.cln-grpc.sh on
```
Do these fail when being run manually?

cstenglein, 2023-02-13:
About the stuck state: I can't even CTRL+C out of it, need to close the connection here.

The above commands work & now the API has no problem communicating with CLN anymore :+1: 

I think  the issue is the stuck state, which provides no feedback & progress. Most likely the grpc plugin install is running in the background. Maybe something is blocking the feedback?

--- 
Small bug:
With the 
`/home/admin/config.scripts/cl-plugin.cln-grpc.sh on` command there is one small error:

```
rm: cannot remove '/home/bitcoin/cl-plugins-enabled/cln-grpc': No such file or directory
```
but it runs through.

openoms, 2023-02-13:
Are you running these on top of the fatpack image (so the install should be already done)?
The rust builds can take quite a while on the Rpi, will add more feedback to it and hide the unnecessary warning.

cstenglein, 2023-02-13:
Yes, used the fatpack (sry for not clarifying).

The rust stuff took ~16m. As I said, usually you get feedback in the installation. What's weird here is that it looks like it's stuck on the part above, but most likely it triggers the grpc installation.
In addition, it's not possible to exit at all, so that's why my guess was a blocking process or similar.

closed 2023-05-10 by rootzoll