raspiblitz/raspiblitz #3678: subcription menu error

openoms, 2023-02-13:
when choosing IP2Tor there is a syntax error:
```
please wait ..
/home/admin/config.scripts/bonus.sphinxrelay.sh: line 135: syntax error near unexpected token `$'"\n    whiptail --title "''
/home/admin/config.scripts/bonus.sphinxrelay.sh: line 135: `    whiptail --title " Warning " \'
```

openoms, 2023-02-13:
Then when running a custom bridge:
```
# running: python /home/admin/config.scripts/blitz.subscriptions.ip2tor.py create-ssh-dialog CUSTOM REDACTED.onion 80
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.ip2tor.py", line 20, in <module>
    from lndlibs import lightning_pb2 as lnrpc
  File "/home/admin/config.scripts/lndlibs/lightning_pb2.py", line 20, in <module>
    _OUTPUTSCRIPTTYPE = DESCRIPTOR.enum_types_by_name['OutputScriptType']
AttributeError: 'NoneType' object has no attribute 'enum_types_by_name'
```

This is on a node which was not reflashed for a few versions,so might just need a custom fix and will need to test on a clean one to confirm.

rootzoll, 2023-02-13:
@openoms the sphix install gets fixed with PR #3681 

The custom IP2Tor problem I was not able to reproduce yet - please share a custom onion address that has this problem if you can reproduce on latest rc build.