raspiblitz/raspiblitz #3679: Proxmox: # FAIL # Couldn't download Wladimir...

Camboa, 2023-02-13:
I tried to install Raspiblitz on Proxmox following the instructions from cercatrova.blog.
Unfortunately there is always an error during the installation:

`*** PREPARING BITCOIN ***
gpg: Verzeichnis `/root/.gnupg' erzeugt
gpg: Die "Keybox" `/root/.gnupg/pubring.kbx' wurde erstellt
gpg: Empfangen vom Schlüsselserver fehlgeschlagen: Server indicated a failure
# FAIL # Couldn't download Wladimir J. van der Laan's PGP pubkey`

rootzoll, 2023-02-13:
@cercatrova21 can you take a look?

cercatrova21, 2023-02-14:
This is not a proxmox problem. This is a general problem in bitcoin.install.sh script. The gpg key from laanwj (71A3 B167 3540 5025 D447 E8F2 7481 0B01 2346 C9A6) is not on the keyserver anymore. We should add another gpg key for verification?

cercatrova21, 2023-02-14:
https://github.com/bitcoin/bitcoin/commit/296e88225096125b08665b97715c5b8ebb1d28ec

cercatrova21, 2023-02-14:
Ok I found the key on the keyring server, but without the spaces. I created a branch with updated bitcoin.install.sh. @Camboa can you try it with the following script?

```
wget https://raw.githubusercontent.com/cercatrova21/raspiblitz/keyserver-fix/build_sdcard.sh
sudo bash build_sdcard.sh
```

Camboa, 2023-02-14:
@cercatrova21 Unfortunately still the same error. I tried it with a new VM.

```
*** PREPARING BITCOIN ***
gpg: Verzeichnis `/root/.gnupg' erzeugt
gpg: Die "Keybox" `/root/.gnupg/pubring.kbx' wurde erstellt
gpg: Empfangen vom Schlüsselserver fehlgeschlagen: Server indicated a failure
# FAIL # Couldn't download Wladimir J. van der Laan's PGP pubkey

```

Camboa, 2023-02-14:
What exactly do you need? If it works for you, I need to check my system.

```
*** Adding deb.torproject.org keyring ***
- OK key added over Tor

*** Adding Tor Sources ***
deb [arch=amd64] tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye main
deb-src [arch=amd64] tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org  bullseye main

- OK sources added

*** Reinstall ***
OK:1 http://deb.debian.org/debian bullseye InRelease                                                                                       
OK:2 http://security.debian.org/debian-security bullseye-security InRelease                                                                
OK:3 http://deb.debian.org/debian bullseye-updates InRelease                                                
Holen:4 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease [3’526 B]
Holen:5 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye/main Sources [1’258 B]
Holen:6 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye/main amd64 Packages [2’434 B]    
Es wurden 7’218 B in 14 s geholt (513 B/s).                                                                                                
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Aktualisierung für 2 Pakete verfügbar. Führen Sie »apt list --upgradable« aus, um sie anzuzeigen.
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Vorgeschlagene Pakete:
  mixmaster torbrowser-launcher socat apparmor-utils
Die folgenden Pakete werden aktualisiert (Upgrade):
  tor
1 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
Es müssen 2’032 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 10.2 kB Plattenplatz zusätzlich benutzt.
Holen:1 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye/main amd64 tor amd64 0.4.7.13-1~d11.bullseye+1 [2’032 kB]
Es wurden 2’032 kB in 41 s geholt (49.6 kB/s).                                                                                             
apt-listchanges: Changelogs werden gelesen...
(Lese Datenbank ... 48328 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../tor_0.4.7.13-1~d11.bullseye+1_amd64.deb ...
Entpacken von tor (0.4.7.13-1~d11.bullseye+1) über (0.4.5.16-1) ...
tor (0.4.7.13-1~d11.bullseye+1) wird eingerichtet ...
Neue Version der Konfigurationsdatei /etc/apparmor.d/abstractions/tor wird installiert ...
Trigger für man-db (2.9.4-2) werden verarbeitet ...
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
apt-transport-tor ist schon die neueste Version (0.5).
gpg ist schon die neueste Version (2.2.27-2+deb11u2).
nyx ist schon die neueste Version (2.1.0-2.1).
obfs4proxy ist schon die neueste Version (0.0.8-1+b6).
python3-stem ist schon die neueste Version (1.8.0-3).
torsocks ist schon die neueste Version (2.3.0-3).
curl ist schon die neueste Version (7.74.0-1.3+deb11u5).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.

*** PREPARING BITCOIN ***
gpg: Verzeichnis `/root/.gnupg' erzeugt
gpg: Die "Keybox" `/root/.gnupg/pubring.kbx' wurde erstellt
gpg: Empfangen vom Schlüsselserver fehlgeschlagen: Server indicated a failure
# FAIL # Couldn't download Wladimir J. van der Laan's PGP pubkey
```

cercatrova21, 2023-02-15:
Do you get an error when you add the key manually?
Try
`gpg --keyserver hkp://keyserver.ubuntu.com --recv-key "71A3B16735405025D447E8F274810B012346C9A6"`
or
`gpg --keyserver hkp://keyserver.ubuntu.com --recv-key "71A3 B167 3540 5025 D447 E8F2 7481 0B01 2346 C9A6"`

openoms, 2023-02-15:
This could be the ISP / DNS erver blocking the keyserver.

Try the alternative:
```
gpg --keyserver hkps://keys.openpgp.org --recv-key "71A3 B167 3540 5025 D447 E8F2 7481 0B01 2346 C9A6"
```
also recommended in: https://github.com/ElementsProject/lightning/blob/master/SECURITY.md - we could change to that.

cercatrova21, 2023-02-15:
@Camboa I changed the keyserver in this branch. You can test it again with:

```
wget https://raw.githubusercontent.com/cercatrova21/raspiblitz/keyserver-fix/build_sdcard.sh
sudo bash build_sdcard.sh
```

Camboa, 2023-02-16:
Still the same error.
I also tried to add the key manually:

```
root@vRaspiblitz:~# gpg --keyserver hkps://keys.openpgp.org --recv-key "71A3 B167 3540 5025 D447 E8F2 7481 0B01 2346 C9A6"
gpg: Verzeichnis `/root/.gnupg' erzeugt
gpg: Die "Keybox" `/root/.gnupg/pubring.kbx' wurde erstellt
gpg: Empfangen vom Schlüsselserver fehlgeschlagen: Server indicated a failure
root@vRaspiblitz:~# gpg --keyserver hkp://keyserver.ubuntu.com --recv-key "71A3 B167 3540 5025 D447 E8F2 7481 0B01 2346 C9A6"
gpg: Empfangen vom Schlüsselserver fehlgeschlagen: Server indicated a failure
root@vRaspiblitz:~# gpg --keyserver hkp://keyserver.ubuntu.com --recv-key "71A3B16735405025D447E8F274810B012346C9A6"
gpg: Empfangen vom Schlüsselserver fehlgeschlagen: Server indicated a failure
```

There must be something wrong on my side. Thanks for your help!

cercatrova21, 2023-02-16:
Yes, looks like you have a good (too good) firewall ;)

Camboa, 2023-02-16:
I found the problem.

In my /etc/resolv.conf file was:
`nameserver 192.168.1.1`

I changed it to:
`nameserver 8.8.8.8`

Now it works! Thanks!

closed 2023-02-16 by Camboa