raspiblitz/raspiblitz #3685: Question on backing up LNbits database

Chris869218, 2023-02-18:
To backup LNbits users (if being used with lnurl-auth and not wanting to risk an account being lost), the move would be to backup /mnt/hdd/app-data/LNBits and to stop the LNBits service before copying the database. To safely stop the service, and restart after, would that be: 'sudo systemctl disable lnbits' and then 'sudo systemctl enable lnbits'? or would it be 'lnbits.service' instead of 'lnbits'? Or would the command be 'sudo systemctl start lnbits instead' of 'enable'? 

Thanks. Wanted to ask first before breaking anything. Appreciate it.

rootzoll, 2023-03-07:
LNbits could also run with a Postgres data ... so first check if you see a file called `/mnt/hdd/app-data/LNBits/database.sqlite3` then the backup of just the `app-data/LNBits ` directory should be enough.

rootzoll, 2023-03-07:
to stop before backup simply use: `sudo systemctl stop lnbits`

to check if it really is not running use `sudo systemctl status lnbits` and it should show you `Active: inactive (dead)`

to start after backup use: `sudo systemctl start lnbits`

rootzoll, 2023-03-07:
BTW if LNbits runs with Postgres database you dont need to start/stop to create a backup. You while running go MAINMENU > LNBITS and there use the OPTIONS > BACKUP feature. It will create a database dumpo file that you then can use for your backups.

Chris869218, 2023-03-12:
Great, thank you!

closed 2023-03-12 by Chris869218