raspiblitz/raspiblitz #3691: Configure apps to use rpc_cookie authentication

openoms, 2023-02-24:
## Configure apps to use the rpc cookie
basic notes to be followed command by command

## Mempool works ok
## CLN works ok
## LND works ok

## Fulcrum
```
sudo usermod -a -G bitcoin fulcrum
#comment out rpcuser and rpcpassword in fulcrum.conf
sudo sed -i "s/^rpcuser/#rpcuser/g" /home/fulcrum/.fulcrum/fulcrum.conf
sudo sed -i "s/^rpcpassword/#rpcpassword/g" /home/fulcrum/.fulcrum/fulcrum.conf
# add the rpccookie option
if ! grep -q "^rpccookie" /home/fulcrum/.fulcrum/fulcrum.conf; then
  echo "rpccookie=/mnt/hdd/bitcoin/.cookie" | sudo tee -a /home/fulcrum/.fulcrum/fulcrum.conf
else
  sudo sed -i "s/^#rpccookie/rpccookie/g" /home/fulcrum/.fulcrum/fulcrum.conf
fi
# make the .cookie file group readable
sudo chmod 640 /mnt/hdd/bitcoin/.cookie
# restart fulcrum
sudo systemctl restart fulcrum
```

## Electrs
```
# comment out auth="USER:PASSWORD" in the electrs.toml
sudo sed -i "s/^auth/#auth/g" /home/electrs/.electrs/config.toml
# add the elecrs user to the bitcoin group
sudo usermod -a -G bitcoin electrs
# symlink the cookie file to the electrs home dir
sudo -u electrs mkdir /home/electrs/.bitcoin
sudo ln -s /mnt/hdd/bitcoin/.cookie /home/electrs/.bitcoin/.cookie
# make the .cookie file group readable
sudo chmod 640 /mnt/hdd/bitcoin/.cookie
# restart electrs
sudo systemctl restart electrs
```
## Specter
```
# this process is to be simplified
# add the user to the bitcoin group
sudo usermod -a -G bitcoin specter
# symlink the cookie file to the home dir
sudo -u specter mkdir /home/specter/.bitcoin
sudo ln -s /mnt/hdd/bitcoin/.cookie /home/specter/.bitcoin/.cookie

# move /home/specter/.specter/config.json
sudo mv /home/specter/.specter/config.json /home/specter/.specter/config.json.bak
## move /home/specter/.specter/nodes/default.json
sudo mv /home/specter/.specter/nodes/default.json /home/specter/.specter/nodes/default.json.bak
# move /home/specter/.specter/nodes/raspiblitz_mainnet.json
sudo mv /home/specter/.specter/nodes/raspiblitz_mainnet.json /home/specter/.specter/nodes/raspiblitz_mainnet.json.bak

# make new specter config
proxy="socks5h://localhost:9050"
torOnly="true"
tor_control_port="9051"
chain="main"
cat > /home/admin/config.json <<EOF
{
    "auth": {
        "method": "none",
        "password_min_chars": 6,
        "rate_limit": 10,
        "registration_link_timeout": 1
    },
    "active_node_alias": "raspiblitz_${chain}net",
    "proxy_url": "${proxy}",
    "only_tor": "${torOnly}",
    "tor_control_port": "${tor_control_port}",
    "tor_status": true,
    "hwi_bridge_url": "/hwi/api/"
}
EOF
sudo mkdir -p /home/specter/.specter/nodes
sudo mv /home/admin/config.json /home/specter/.specter/config.json
sudo chown -RL specter:specter /home/specter/

sudo systemctl stop specter
# here I had an erro where needed to move the specter wallets folder
# not sure if this is reoccuring
sudo -u bitcoin cp -r /mnt/hdd/bitcoin/specter /mnt/hdd/bitcoin/specternewfolder

# restart specter
sudo systemctl restart specter

# open the Specter WebUI and set a password ASAP
```

mitohund, 2023-03-01:
Since I installed ord according to [your beautiful instructions](https://github.com/rootzoll/raspiblitz/issues/3690), my LND stopped working. Which is fine with me - I'm not using Ligthning. I would actually like to turn LND off so it stops asking me to regularly type in my password (which it does quite frequently). Is that possible?

mitohund, 2023-03-01:
For anybody wanting to turn off Lightning on their RB:

`nano /mnt/hdd/raspiblitz.conf`
add or edit to: `lightning=off`