raspiblitz/raspiblitz #3694: No Signal - Touchscreen Display RPi4b/128gb

AlcSndrsn, 2023-02-27:
Delete or move if this isn't the right place.....

I've set up a raspiblitz before using ssh, but this time around I heard about the LCD screens. 

I initially purchased: [Raspberry Pi Official 7 Inch Touch Screen](https://www.amazon.com/dp/B073S3LQ6Q?psc=1&ref=ppx_yo2ov_dt_b_product_details). This didn't work via ribbon cable (ssh'd and activated tft). I figured I had made a mistake and saw some other issues in git re: issues with _"larger"_ displays. 

So I then purchased [kuman TFT Touch Screen+case, 3.5 Inch TFT LCD Display Monitor with Protective Case Support All Raspberry Pi System](https://www.amazon.com/dp/B075JFHLQD?psc=1&ref=ppx_yo2ov_dt_b_product_details), which I believe is on your alternative display's list in the Basic Setup Readme. This plugs directly into the pi using the headers and there's still 'no signal' appearing. 

Is my pi the issue? my build? my screen(s). 

Any help would be greatly appreciated, I can provide missing information as best as possible.

Thanks again. and feel free to delete if this isn't wanted.

rootzoll, 2023-03-07:
kuman TFT Touch you have seems to be a HDMI display (as linked to amazon) uses HDMI for video signal .. thats not the standard Touchdisplay we recommend and develop for.  There is a Kuman working just thru the GPIO pins like this one: https://www.amazon.de/Kuman-Protective-Resolution-Raspberry-Bildschirm/dp/B07KSHVC3H

AlcSndrsn, 2023-03-07:
First, I didn't really expect to receive a response, so thanks for that in and of itself. 

I've yet to try joksbit idea....
But I am very open to have picked the wrong or incompatible lcd display. 

Rootzool, huge fan. The OG Kuman screen above and the one you shared, both say they have HDMI as their output. Both my Kuman and the one you suggested can be setup either through _GPIO or via HDMI_ (or both). 

If you have 1 touch display to choose for a rpi **4b** - what is it and why?

Thanks so much in advance.

rootzoll, 2023-03-08:
> Rootzool, huge fan. The OG Kuman screen above and the one you shared, both say they have HDMI as their output. Both my Kuman and the one you suggested can be setup either through GPIO or via HDMI (or both).

I have no real experience with the Kuman LCD I linked .. but as far as I read ..
```
The 3.5 in touch screen can support two screens simultaneous display(The Raspberry Pi can be connected to this screen and another HDMI screen at the same time). 
``
... they mean that this model does not use the HDMI port of the raspi and an additional screen can still be connected thru the original HDMI port of the pi. The one that you have looks like it uses the HDMI for video signal (thats why a HDMI bridge plug is included) and it uses the GPIO ports for the touch control .. as far as I ca see.

> If you have 1 touch display to choose for a rpi 4b - what is it and why?

With all LCD screen models dont use the ones that have an HDMI port/connector - what you need is a 3.5 inch LCD screen model (resolution of 480×320) that connects only thru the GPIO ports (SPI) and has an XPT2046 touch controller.

The reference model is the Waveshare 3.5 (c) https://www.waveshare.com/3.5inch-rpi-lcd-c.htm

Its needed because we use the following project to drive the LCD:
https://github.com/tux1c/wavesharelcd-64bit-rpi/blob/master/README.md