raspiblitz/raspiblitz #3698: Keeps asking for PASSWORD C

BitBoxer75, 2023-03-05:
The issue is that I have to put the password c in several times, so after one to two minutes and after a restart of the node I stuck with this screen and have to add the password c repeatedly in.

[image]

When I exit this screen and call from the main menu the Infoscreen shows this.

[image]

Some background, the issue most likely originates back when I had an GetUmbrel node. At a point in time, my LND Node was frozen. As I doesn't found a solution for that issue in the Umbrel community, I decided to move to a Raspiblitz. After migration, the issue persisted, and I moved on, as the Bitcoin FullNode was running. Some weeks ago I decided to update the Raspiblitz to version 1.8, with the slight hope the issue will solve automatically, but it hasn't, so it's time for solving this.

I have read the issue # 336 and checked both the bitcoin.conf and lnd.conf, both have the same RPC password.

Here is the storage of the debug file.

[https://l.termbin.com/lcfz1](https://l.termbin.com/lcfz1)

Many thanks in advance for your help.

rootzoll, 2023-03-07:
Normally when you get asked for passwordC on repeat it means either the passwordC you entered was wrong OR the lightning client failed and restarted so that the database needs to get unlocked again. 

I your case it looks like the last one - because in your debug file your LND is just active running for 18 seconds ... it just freshly restarted.

Also in the LND error log there is some error report:
```
*** LAST LND (MAINNET) ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Journal begins at Sat 2023-03-04 11:57:06 GMT, ends at Sun 2023-03-05 12:27:30 GMT. --
Mar 05 12:26:08 raspberrypi lnd[736622]: github.com/btcsuite/btcwallet/wallet.(*Wallet).handleChainNotifications(0x400aac8000)
Mar 05 12:26:08 raspberrypi lnd[736622]:         github.com/btcsuite/btcwallet@v0.16.5/wallet/chainntfns.go:117 +0xe08
Mar 05 12:26:08 raspberrypi lnd[736622]: created by github.com/btcsuite/btcwallet/wallet.(*Wallet).SynchronizeRPC
Mar 05 12:26:08 raspberrypi lnd[736622]:         github.com/btcsuite/btcwallet@v0.16.5/wallet/wallet.go:213 +0x3c8
Mar 05 12:26:08 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Mar 05 12:26:08 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Mar 05 12:26:08 raspberrypi systemd[1]: lnd.service: Consumed 56.254s CPU time.
```

Its not listed there in full - so it would be helpfull if you can post longer listing of that error log. EXIT to the command line (terminal) and call command `sudo journalctl -u lnd -b --no-pager -n64` and just copy paste the result here.

BitBoxer75, 2023-03-08:
Many thanks for pick my issue up. Here is the result (not much longer, hope that helps) from the command: sudo journalctl -u lnd -b --no-pager -n64

-- Journal begins at Wed 2023-03-08 06:08:27 GMT, ends at Wed 2023-03-08 19:30:38 GMT. --
Mar 08 19:30:31 raspberrypi lnd[3832180]: panic: unable to synchronize wallet to chain: failed to fetch block hash for height 711828: block not found
Mar 08 19:30:31 raspberrypi lnd[3832180]: goroutine 169295 [running]:
Mar 08 19:30:31 raspberrypi lnd[3832180]: github.com/btcsuite/btcwallet/wallet.(*Wallet).handleChainNotifications(0x4009cd0000)
Mar 08 19:30:31 raspberrypi lnd[3832180]:         github.com/btcsuite/btcwallet@v0.16.5/wallet/chainntfns.go:117 +0xe08
Mar 08 19:30:31 raspberrypi lnd[3832180]: created by github.com/btcsuite/btcwallet/wallet.(*Wallet).SynchronizeRPC
Mar 08 19:30:31 raspberrypi lnd[3832180]:         github.com/btcsuite/btcwallet@v0.16.5/wallet/wallet.go:213 +0x3c8
Mar 08 19:30:31 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Mar 08 19:30:31 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Mar 08 19:30:31 raspberrypi systemd[1]: lnd.service: Consumed 14min 20.599s CPU time.

BitBoxer75, 2023-03-13:
Many thanks for the answer. I had used on Raspiblitz 1.8 in the menu the `repair-option` → `reset-chain` and downloaded the blockchain completely before I raised this issue. Will be there a difference on the new v1.9.0?

rootzoll, 2023-03-14:
The v1.9.0 will have some more repair options like recreating indexes but reset-chain is the same as before.

If a complete blockchain reset/redownload didnt solved your issue - maybe see if someone from Lightning Labs can give you more details what is causing this the `panic: unable to synchronize wallet to chain: failed to fetch block hash for height 711828: block not found` error log.

Best would be to try the LND Dev Slack and ask there: https://lightning.engineering/slack.html

If you find out more details please repost them here so that maybe in the future we can develop an repair option to fix this problem.

BitBoxer75, 2023-03-26:
I have raised a question in the suggested slack channel, [https://lightningcommunity.slack.com/archives/C6AFCN3KL/p1679843977894729](https://lightningcommunity.slack.com/archives/C6AFCN3KL/p1679843977894729), and will post the results here.

Many thanks for the support sofar.

billgre, 2023-03-29:
I had the same issue a month ago. I left tor network , and works. But that is not a solution. I follow the thread , please post the answer here. Thanks