raspiblitz/raspiblitz #3700: UNABLE TO MANAGE LND WALLET : channels inactive on RTL but active on 1ml.com site

joksbit, 2023-03-06:
I started a Ligthning network 9 months ago, for educational purpose with teenagers. I was able to fund the LND Wallet and open channels. The problem started after an update and a prolonged synchronizing of the LN. The withdrawal of funds becomes impossible. I am not a techie and don't know how to go about the issue. 

I have tried various options to closed the channels but they remain active on Iml.com . Whereas on the RTL they all appear inactive:
Ride The Lightning - jokscryptos
Active  0 ;   Pending 0;  Inactive 7: Closing 0

My node's URI is
`03443d4d2248e25ed0bc726aaba3a9a858561f452f62e3c3baa36abe5d99e74d65@hth3b3nfasox6r5qhj6ih53yixbag53yl2jxreeeccplfn4jl7fqhcyd.onion:9735`

I tried to make the back up and saved the backup file to used in restore. Trying to force all channnels to close and release the funnds, it does not work either. 

How do I get access to control the LND again?

rootzoll, 2023-03-08:
Have you tried also the CLOSEALL option in the raspiblitz SSH menus under MAINMENU > LND?

cstenglein, 2023-06-07:
@joksbit  is the issue still relevant?

closing for now, if it still relevant, please re-open the issue. :)

closed 2023-06-07 by cstenglein