raspiblitz/raspiblitz #3705: Several (solved) issues on installing RTL [CLN, dev-branch]

Surenic, 2023-03-10:
Don't know if this helps, not yet a github pro t write a PR but solved some issues (for me) with the RTL and cl.rest scripts.

I was trying to update RTL to 13.6 on my proxmox debian 11 VM

First `cl.rest.sh` off sent an error on `mkdir` and `cp` the directory, which has already been there (started on 1.9rc1)
https://github.com/rootzoll/raspiblitz/blob/6a677a6baacc98a89d99bf93cf9e7053dabfebd5/home.admin/config.scripts/cl.rest.sh#L166

I had to delete the directory manually.

Further the RTL script was not able to replace my RTL 13.0beta installation. I had to remove `/home/rtl/RTL` as well.

Lastly I had an error on `npm install --omit=dev`
https://github.com/rootzoll/raspiblitz/blob/6a677a6baacc98a89d99bf93cf9e7053dabfebd5/home.admin/config.scripts/bonus.rtl.sh#L165

I needed to add `--legacy-peer-deps` as provided [here](https://github.com/Ride-The-Lightning/RTL/blob/master/.github/docs/Core_lightning_setup.md#error-on-npm-install).

Removing c-lightning-REST and RTL again, calling the modified script made RTL 13.6 install eventually.

Hope this helps

rootzoll, 2023-03-13:
👀 @openoms

openoms, 2023-04-05:
Thanks for these @Surenic!
`cl.rest.sh off` has a `purge` option as well to remove everythin, now added to be used with `bonus.rtl.sh off purge`

closed 2023-06-07 by cstenglein