raspiblitz/raspiblitz #3713: Force close with several nodes

puehri, 2023-03-19:
Hi,

it is currently the second time, this happens. I don't know why or how.
Sometimes my node automatically closes channels (Today were 4 channels closed. Only 3 in the screenshot, the other one already closed).

Is there a setting i don't know? Or might this be a bug? 
Has anyone else experienced this?

I am using the currently released version of RaspiBlitz 1.8.

[image]

rootzoll, 2023-03-20:
There is no special feature or setting in raspiblitz to force close channels automatically ... best would be (if you havent already) to ask in the Telegram groups if others saw similar things (links listed at https://raspiblitz.org) or ask in other lnd or cln user groups.

puehri, 2023-04-23:
Did not happen again. -> Closed

closed 2023-04-23 by puehri