raspiblitz/raspiblitz #3714: debug.log no such file

al3xisxyz, 2023-03-20:
hi,
My debug.log keeps disappearing. 
It was working fine but after a restart when it is not present anymore and in the info screen I get this error message:
`tail: cannot open file '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory`

I got this issue in another device too, and I even tried creating the file myself but it is not getting populated with logs.

It has happened with v1.8 and v1.9 now.

rootzoll, 2023-03-20:
The raspiblitz is deleting the bitcoin `debug.log` on every bootup here in the script:
https://github.com/rootzoll/raspiblitz/blob/1f06f90b95da39f1593b4a9a483bf85f244b4fac/home.admin/_bootstrap.sh#L876

For bitcoind this is is OK because it will recreate a new log file. I will check where the infoscreen has problems with accessing it.

al3xisxyz, 2023-03-20:
> The raspiblitz is deleting the bitcoin `debug.log` on every bootup here in the script: [...]

Ah great, good to see that it is deleted on restart by design. But in my case when I check in the folder after bitcoind has started then I don't see the file. So it is not recreated for some reason. I manually made the file once with the user bitcoin, but it didn't generate any logs inside it later.

rootzoll, 2023-03-20:
Not sure why it not gets recreated on your machine. The location of the debug.log file is directly set with the systemd service ... can you check if you get a similar output with command `sudo systemctl status bitcoind`

```
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-02-27 18:22:54 GMT; 3 weeks 0 days ago
   Main PID: 8098 (bitcoind)
      Tasks: 19 (limit: 9353)
        CPU: 3d 15h 47min 57.163s
     CGroup: /system.slice/bitcoind.service
             └─8098 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log
```

al3xisxyz, 2023-03-20:
> 

Yeah exactly the same including all the right paths... 
I have Raspiblitz on a VM and on a raspberry pi 4, both same behavior...

al3xisxyz, 2023-03-22:
> Not sure why it not gets recreated on your machine. The location of the debug.log file is directly set with the systemd service ... can you check if you get a similar output with command `sudo [...]

```
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-03-19 17:17:47 CET; 2 days ago
   Main PID: 2042 (bitcoind)
      Tasks: 18 (limit: 9505)
     Memory: 3.2G
        CPU: 4h 28min 813ms
     CGroup: /system.slice/bitcoind.service
             └─2042 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Warning: journal has been rotated since unit was started, output may be incomplete.

```

rootzoll, 2023-03-23:
Hmm .. so if the bitcoind is told directly to create a `degub.log` and it doesnt get created - I am an bit clueless why. Maybe File permissions, etc .. one last way would be to monitor if there is anything from bitcoind complaining on creating logs when you start it manually - you can try

```
sudo systemctl stop bitcoind
```

and then

```
sudo -u bitcoin /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log
```

If no error message comes - restart node after this.

al3xisxyz, 2023-03-23:
> `sudo -u bitcoin /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log`

Everything worked without any error message and the log file was created and started filling up with logs. But when I restarted the node I am back to the same issue as before with no log file... Weird, everything else seems to be working so maybe I shouldn't worry about it?

al3xisxyz, 2023-04-18:
> Ok so the problem seems when you run bitcoin as systemd - but not when you start manual. Can you post the content of your `bitcoind.service` file? Use `sudo cat /etc/systemd/system/bitcoind.service`

Ok here is the output:
```
sudo cat /etc/systemd/system/bitcoind.service
# RaspiBlitz: systemd unit for bitcoind
# based  on https://github.com/bitcoin/bitcoin/blob/master/contrib/init/bitcoind.service
[Unit]
Description=Bitcoin daemon

After=network-online.target
Wants=network-online.target

# for use with sendmail alert
#OnFailure=systemd-sendmail@%n

[Service]
ExecStartPre=-/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED
ExecStart=/usr/local/bin/bitcoind -daemonwait \
                                  -conf=/mnt/hdd/bitcoin/bitcoin.conf \
                                  -datadir=/mnt/hdd/bitcoin \
                                  -debuglogfile=/mnt/hdd/bitcoin/debug.log

# Make sure the config directory is readable by the service user
PermissionsStartOnly=true
ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin

# Process management
####################
Type=forking
Restart=on-failure
TimeoutStartSec=infinity
TimeoutStopSec=600

# Directory creation and permissions
####################################
# Run as bitcoin:bitcoin
User=bitcoin
Group=bitcoin

StandardOutput=null
StandardError=journal

# Hardening measures
####################
# Provide a private /tmp and /var/tmp.
PrivateTmp=true
# Mount /usr, /boot/ and /etc read-only for the process.
ProtectSystem=full
# Deny access to /home, /root and /run/user
ProtectHome=true
# Disallow the process and all of its children to gain
# new privileges through execve().
NoNewPrivileges=true
# Use a new /dev namespace only populated with API pseudo devices
# such as /dev/null, /dev/zero and /dev/random.
PrivateDevices=true
# Deny the creation of writable and executable memory mappings.
MemoryDenyWriteExecute=true

[Install]
WantedBy=multi-user.target
```

rootzoll, 2023-04-18:
Looks OK so far. Kind of running out of ideas what else to check for.

cstenglein, 2023-06-07:
@al3xisxyz  is this still an issue with 1.9.0?

Closing this for now, please re-open if it still persists.

grnqrtr, 2023-06-26:
I'm having this same issue on 1.9.0. I confirmed that when I start bitcoind mannually, the debug.log is created, but when it is ran with systemd no debug.log is created.

rootzoll, 2023-07-31:
see #4003 and the open PR #4051

closed 2023-09-14 by rootzoll