raspiblitz/raspiblitz #3719: Lost all channels... Can't recover funds

homenode80, 2023-03-23:
Hi guys,

Since I updated a few months ago to Taproot upgrade, I lost all my channels on my Raspiblitz.

I tried to recover but i have no rescue file apparently and only onchain funds are showing.

With the recovery file how can I recover my channels and then my BTC?

My pubkey is:

035bddb48a2b891a8e2a704234cb13405d63c62fcd8aa1c6f5ea10c4c027654af7

rootzoll, 2023-03-23:
I dont think it was "taproot" that made all your channel go away. Maybe they got (force) closed by your channel partners? I mean what do you see if you for example check with the App RideTheLightning under Lightning > Channels ? Are there any listed under pending/closed?

As of backup ... did you at any point downloaded a StaticChannelBackup file (its called like channel.backup). Because with just the seed words you have no access to any channel funds.

homenode80, 2023-03-24:
Hi and thanks for your reply.
There are no channels in pending/closed like nothing ever was there...
I just have the seed words no other backup file is available.
Am I screwed...? :(

versieboer, 2023-03-24:
I see no evidence of any channels ever having been built to pubkey 035bddb48a2b891a8e2a704234cb13405d63c62fcd8aa1c6f5ea10c4c027654af7 in any of the LN explorers that I checked.

I am certain that you have the wrong pubkey somehow.  Where did you get that pubkey from?

I did the same thing once when I used the wrong hardware- I tried to set it up on 2 different RasPi's.  Set it up on 1 successfully.  Forgot about it for a few months.  Then figured out that I was looking later at my 2nd Pi that I was trying to get it set up on.

openoms, 2023-03-30:
It is not clear what do you mean by upgrading to Taoroot. Did you only uodate your bitcoind version or what you did exactly?

homenode80, 2023-03-30:
I updated my Raspiblitz to the latest Bitcoin core version after Taproot upgrade (as was explained to do in the Github), once I did that all my channels were lost.

Henkzonfire, 2023-03-31:
Same thing happened to me but thanks to Blackhathacker111 who helped me message him on telegram his telegram username is https://t.me/Blackhathacker111

openoms, 2023-05-01:
You must have done something different  from just upgrading Bitcoin Core.

Lacking the old lnd directory what you need is the channel.backup (SCB) file to recover.

Is 
```
lncli getinfo
```
working?

Did you update LND to a recent version? It would not even work under v0.15.4. Best to be on v0.16.2. You can upgrade in the menu.

DrrDino, 2023-06-03:
> I only have one wallet that was setup when I installed Raspiblitz, I'm sure it is the one because there are still 100,000 sats onchain which correspond to that wallet. I recovered it after I upgraded to Taproot with the only seedphrase that I have...
> I don't understand how the upgrade wiped out all the channel data but I can assure you that I had at least 0.1 BTC alltogether in those channels.

Same thing happened to me but thanks to Blackhathacker111 who helped me recover all my funds,message him on telegram his telegram username is  https://t.me/Blackhathacker111