raspiblitz/raspiblitz #3721: Note hangs regularly after update to 1.9.0rc2

JoFi7, 2023-03-25:
After updating to 1.9.0rc2 the note hangs after a few days. The SSD no longer flashes but lights up continuously. The display is also frozen. The note can no longer be reached via the terminal. So far I have only been able to unplug the power cord. When the Note then reboots it will function normally again.

openoms, 2023-03-30:
This suggests a hardware error usually with the SSD power supply. It might worth to change the USB adapter. Which one are using?
My well working setup uses an X825 SATA uSB adapter for the SSD with an official RPi power supply.

JoFi7, 2023-03-31:
Hi, thank you very much for your reply. My setup is from Fulmo. I have the same USB adapter for SSD. I went back to the last version 1.8.0. The SSD has been running again for three days without any problems. I'll watch it for a week and report back. Greetings Joerg

JoFi7, 2023-03-31:
....... which I forgot to mention. Under 1.9.0, the node was frozen when the SSD stopped blinking, the node could no longer be reached via the SSH terminal.

JoFi7, 2023-04-07:
Hi, the node runs again with the software 1.8.0 without any problems. For me it was probably actually because of the software 1.9.0. Greetings Joerg

closed 2023-04-07 by JoFi7