raspiblitz/raspiblitz #3724: state(error) message(cl-wallet-new-nowallet)

dipunm, 2023-03-27:
After trying to set up my node using build_sdcard.sh, I was greeted with the following error:

(I chose CLN as lightning implementation)

```
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(cl-wallet-new-nowallet)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
```

It looks like it was really the end of the provisioning too which is a shame, I should have just told it to run without any lightning implementation.

I re-ran the command that I think should have prevented this error and I see:
```
sudo /home/admin/config.scripts/cl.hsmtool.sh new-force mainnet
# Running 'cl.hsmtool.sh new-force mainnet'
seedwords='cinnamon layer master van sadness icon sweet report fantasy flash ugly horse butter awkward educate find satoshi later unfair abuse update muffin equal pistol'
seedwords6x4='1:cinnamon  2:layer     3:master    4:van       5:sadness   6:icon      
7:sweet     8:report    9:fantasy   10:flash    11:ugly     12:horse    
13:butter   14:awkward  15:educate  16:find     17:satoshi  18:later    
19:unfair   20:abuse    21:update   22:muffin   23:equal    24:pistol   '

# This file was placed by cl.hsmtool.sh
# Contains the seed words from which the hsm_secret in the same directory was generated from
seedwords='cinnamon layer master van sadness icon sweet report fantasy flash ugly horse butter awkward educate find satoshi later unfair abuse update muffin equal pistol'
seedwords6x4='1:cinnamon  2:layer     3:master    4:van       5:sadness   6:icon      
7:sweet     8:report    9:fantasy   10:flash    11:ugly     12:horse    
13:butter   14:awkward  15:educate  16:find     17:satoshi  18:later    
19:unfair   20:abuse    21:update   22:muffin   23:equal    24:pistol   '
# Will be removed safely when the hsm_secret is encrypted.

sudo: lightning-hsmtool: command not found
# Re-init the backup plugin with the new wallet
HEAD is now at 3da5778 explain that plugin can also be added to the config
# Checking dependencies
PATH=$PATH:/home/bitcoin/.local/bin
# Stop the lightningd.service
# Backup the existing old backup on the SDcard
/home/bitcoin/.lightning/bitcoin/backup.lock
# Initialize the backup plugin
Database does not exist yet, created an empty backup file
Initialized backup backend file:///home/bitcoin/lightningd.sqlite3.backup, you can now start c-lightning
# The crontab for admin now is:
@weekly /home/admin/config.scripts/cl-plugin.backup.sh backup-compact mainnet
```

Note very specifically: `sudo: lightning-hsmtool: command not found`

rootzoll, 2023-03-28:
To prevent missunderstandings: When ecaxtly did you used "build_sdcard.sh"? Did you used the sd card image download?

dipunm, 2023-03-28:
I ran it yesterday evening.

I usually run it with these flags though: `-w GB -d headless -f 0`

Also, I have now set up my node without any lightning implementation. I then proceeded to try to install clightning and noticed this:

```
sudo: /usr/local/bin/lightningd: command not found
# The installed Core Lightning version is: 
```

openoms, 2023-03-30:
Which branch did you run from?

Please test the dev branch.

dipunm, 2023-04-04:
I ran from the default branch (I used the wget script on the readme here on github)

I should note that I think my setup THINKS clightning is installed. It is visible in the menu.

The steps I think I should run:

1. run `sudo ./config.scripts/cl.install.sh off` (looks like the binaries were never installed so `purge` is unnecessary)
2. use menu to upgrade to `dev` branch
3. run `sudo ./config.scripts/cl.install on`

Is that a good course of action?

openoms, 2023-04-04:
>run sudo ./config.scripts/cl.install.sh off (looks like the binaries were never installed so purge is unnecessary)
    use menu to upgrade to dev branch
    run sudo ./config.scripts/cl.install on

@dipunm yes, that that should work

dipunm, 2023-04-04:
```bash
admin@192.168.8.161:~ ₿ ./config.scripts/cl.install.sh off
# Running: 'cl.install.sh off'
# Using the settings for: bitcoin mainnet
# Removing the lightningd.service
Removed /etc/systemd/system/multi-user.target.wants/lightningd.service.
# Removing the aliases
```

No logs to show for the switching to dev branch as the menu cleared the output :)

running `sudo ./config.scripts/cl.install.sh on`

Noticed a pause in operation at this point:
```bash
- OK the installation of Core Lightning vv23.02.2 is successful
# Make sure bitcoin is in the debian-tor group
# Add plugin-dir: /home/bitcoin/cl-plugins-enabled
# Add plugin-dir: /home/bitcoin/cl-plugins-available
# Store the lightning data in /mnt/hdd/app-data/.lightning
# Symlink to /home/bitcoin/
/home/bitcoin/.lightning/config
# The file /home/bitcoin/.lightning/config is already present
ls: cannot access '/home/bitcoin/.lightning/bitcoin/hsm_secret': No such file or directory
Create a wallet from seedwords for mainnet
```

Looks like it completed well this time:
```bash
# Adding aliases
alias lightning-cli="sudo -u bitcoin /usr/local/bin/lightning-cli --conf=/home/bitcoin/.lightning/config"
alias cl="sudo -u bitcoin /usr/local/bin/lightning-cli --conf=/home/bitcoin/.lightning/config"
alias cllog="sudo tail -n 30 -f /home/bitcoin/.lightning/bitcoin/cl.log"
alias clconf="sudo nano /home/bitcoin/.lightning/config"

# The installed Core Lightning version is: v23.02.2

# To activate the aliases reopen the terminal or use:
source ~/_aliases
# Monitor the lightningd with:
sudo journalctl -fu lightningd
sudo systemctl status lightningd
# logs:
sudo tail -f /home/bitcoin/.lightning/bitcoin/cl.log
# for the command line options use
lightning-cli help

# CL is now the default lightning implementation
```

What does this tell us? Should I be running my node using the dev branch?

dipunm, 2023-04-19:
Since I am re-setting up my node in https://github.com/rootzoll/raspiblitz/issues/3763, I decided to methodically run the cl.install script by hand.

Immediately, I see that we are stuck on `v0.11.2` which is also true for the `dev` branch.

Is there a reason for this? i.e. compatibility?

dipunm, 2023-04-20:
I used the latest version of CLightning (23.02.2) and it installed successfully, however the next line where we install the cln-grpc failed at the `cargo build` line:

```bash
> sudo -u bitcoin /home/bitcoin/.cargo/bin/cargo build \
     --target-dir /home/bitcoin/cl-plugins-available/cln-grpc
.....
.....
.....

error: failed to run custom build command for `cln-grpc v0.1.2 (/home/bitcoin/lightning/cln-grpc)`

Caused by:
  process didn't exit successfully: `/home/bitcoin/cl-plugins-available/cln-grpc/debug/build/cln-grpc-efb2cb4e3defe1e4/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=proto/node.proto
  cargo:rerun-if-changed=proto

  --- stderr
  thread 'main' panicked at 'Could not find `protoc` installation and this build crate cannot proceed without
      this knowledge. If `protoc` is installed and this crate had trouble finding
      it, you can set the `PROTOC` environment variable with the specific path to your
      installed `protoc` binary.If you're on debian, try `apt-get install protobuf-compiler` or download it from https://github.com/protocolbuffers/protobuf/releases

  For more information: https://docs.rs/prost-build/#sourcing-protoc
  ', /home/bitcoin/.cargo/registry/src/github.com-1ecc6299db9ec823/prost-build-0.11.5/src/lib.rs:1295:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
```

closed 2024-10-01 by rootzoll