raspiblitz/raspiblitz #3726: I'm having problems with my Core Lightning wallet.

JoFi7, 2023-03-31:
Hi, I'm having problems with my Core Lightning wallet. When I try to open my wallet on RTL I get: Error 503 Connect Econnrefused . When I then look in Core Lightning Options under Summery: "Information about this Node", this message keeps popping up: "lightning-cli: Connecting to 'lightning-rpc': Connection refused". I don't know what to do and hope for help. Greetings Joerg

-  lightning-cli: Connecting to 'lightning-rpc': Connection refused
# Starting the summary plugin
--- Logging error ---
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 177, in emit
    self.console.print(renderable, overflow="ignore", crop=False, style=style)
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1752, in print
    extend(render(renderable, render_options))
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1390, in render
    for render_output in iter_render:
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 134, in __rich_console__
    for line in lines:
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/segment.py", line 245, in split_lines
    for segment in segments:
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1368, in render
    renderable = rich_cast(renderable)
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/protocol.py", line 36, in rich_cast
    renderable = cast_method()
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/self_outdated_check.py", line 130, in __rich__
    pip_cmd = get_best_invocation_for_this_pip()
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/entrypoints.py", line 58, in get_best_invocation_for_this_pip
    if found_executable and os.path.samefile(
  File "/usr/lib/python3.9/genericpath.py", line 101, in samefile
    s2 = os.stat(f2)
FileNotFoundError: [Errno 2] No such file or directory: '/usr/bin/pip3.9'
Call stack:
  File "/usr/local/bin/pip", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main.py", line 70, in main
    return command.main(cmd_args)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 101, in main
    return self._main(args)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 223, in _main
    self.handle_pip_version_check(options)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/req_command.py", line 148, in handle_pip_version_check
    pip_self_version_check(session, options)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/self_outdated_check.py", line 237, in pip_self_version_check
    logger.info("[present-rich] %s", upgrade_prompt)
  File "/usr/lib/python3.9/logging/__init__.py", line 1442, in info
    self._log(INFO, msg, args, **kwargs)
  File "/usr/lib/python3.9/logging/__init__.py", line 1585, in _log
    self.handle(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 1595, in handle
    self.callHandlers(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 1657, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 948, in handle
    self.emit(record)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 179, in emit
    self.handleError(record)
Message: '[present-rich] %s'
Arguments: (UpgradePrompt(old='22.1.2', new='23.0.1'),)
--- Logging error ---
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 177, in emit
    self.console.print(renderable, overflow="ignore", crop=False, style=style)
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1752, in print
    extend(render(renderable, render_options))
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1390, in render
    for render_output in iter_render:
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 134, in __rich_console__
    for line in lines:
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/segment.py", line 245, in split_lines
    for segment in segments:
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1368, in render
    renderable = rich_cast(renderable)
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/protocol.py", line 36, in rich_cast
    renderable = cast_method()
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/self_outdated_check.py", line 130, in __rich__
    pip_cmd = get_best_invocation_for_this_pip()
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/entrypoints.py", line 58, in get_best_invocation_for_this_pip
    if found_executable and os.path.samefile(
  File "/usr/lib/python3.9/genericpath.py", line 101, in samefile
    s2 = os.stat(f2)
FileNotFoundError: [Errno 2] No such file or directory: '/usr/bin/pip3.9'
Call stack:
  File "/usr/local/bin/pip", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main.py", line 70, in main
    return command.main(cmd_args)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 101, in main
    return self._main(args)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 223, in _main
    self.handle_pip_version_check(options)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/req_command.py", line 148, in handle_pip_version_check
    pip_self_version_check(session, options)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/self_outdated_check.py", line 237, in pip_self_version_check
    logger.info("[present-rich] %s", upgrade_prompt)
  File "/usr/lib/python3.9/logging/__init__.py", line 1442, in info
    self._log(INFO, msg, args, **kwargs)
  File "/usr/lib/python3.9/logging/__init__.py", line 1585, in _log
    self.handle(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 1595, in handle
    self.callHandlers(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 1657, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 948, in handle
    self.emit(record)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 179, in emit
    self.handleError(record)
Message: '[present-rich] %s'
Arguments: (UpgradePrompt(old='22.1.2', new='23.0.1'),)
lightning-cli: Connecting to 'lightning-rpc': Connection refused

Node URI:
@:

# Running:
lightning-cli -H summary

lightning-cli: Connecting to 'lightning-rpc': Connection refused

Press ENTER to return to main menu.

JoFi7, 2023-04-01:
Good morning I read the logs according to your instructions. Can you read something from there?

Apr 01 06:19:21 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2799.
Apr 01 06:19:21 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
Apr 01 06:19:21 raspberrypi systemd[1]: lightningd.service: Consumed 5.065s CPU time.
Apr 01 06:19:21 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Apr 01 06:19:22 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
Apr 01 06:19:44 raspberrypi sh[815785]: lightningd: /home/bitcoin/.lightning/config line 20: watchtower-proxy=127.0.0.1:9050: unrecognized option
Apr 01 06:19:44 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 06:19:44 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
Apr 01 06:19:44 raspberrypi systemd[1]: lightningd.service: Consumed 5.007s CPU time.
Apr 01 06:20:14 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2800.
Apr 01 06:20:14 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
Apr 01 06:20:14 raspberrypi systemd[1]: lightningd.service: Consumed 5.007s CPU time.
Apr 01 06:20:14 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Apr 01 06:20:14 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
Apr 01 06:20:36 raspberrypi sh[821993]: lightningd: /home/bitcoin/.lightning/config line 20: watchtower-proxy=127.0.0.1:9050: unrecognized option
Apr 01 06:20:36 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 06:20:36 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
Apr 01 06:20:36 raspberrypi systemd[1]: lightningd.service: Consumed 4.833s CPU time.
Apr 01 06:21:06 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2801.
Apr 01 06:21:06 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
Apr 01 06:21:06 raspberrypi systemd[1]: lightningd.service: Consumed 4.833s CPU time.
Apr 01 06:21:06 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Apr 01 06:21:06 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
Apr 01 06:21:28 raspberrypi sh[827426]: lightningd: /home/bitcoin/.lightning/config line 20: watchtower-proxy=127.0.0.1:9050: unrecognized option
Apr 01 06:21:28 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 06:21:28 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
Apr 01 06:21:28 raspberrypi systemd[1]: lightningd.service: Consumed 5.007s CPU time.
Apr 01 06:21:58 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2802.
Apr 01 06:21:58 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
Apr 01 06:21:58 raspberrypi systemd[1]: lightningd.service: Consumed 5.007s CPU time.
Apr 01 06:21:58 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Apr 01 06:21:59 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
Apr 01 06:22:19 raspberrypi sh[832949]: lightningd: /home/bitcoin/.lightning/config line 20: watchtower-proxy=127.0.0.1:9050: unrecognized option
Apr 01 06:22:19 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 06:22:19 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
Apr 01 06:22:19 raspberrypi systemd[1]: lightningd.service: Consumed 5.044s CPU time.
Apr 01 06:22:50 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2803.
Apr 01 06:22:50 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
Apr 01 06:22:50 raspberrypi systemd[1]: lightningd.service: Consumed 5.044s CPU time.
Apr 01 06:22:50 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Apr 01 06:22:50 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
Apr 01 06:23:13 raspberrypi sh[838110]: lightningd: /home/bitcoin/.lightning/config line 20: watchtower-proxy=127.0.0.1:9050: unrecognized option
Apr 01 06:23:13 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 06:23:13 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
Apr 01 06:23:13 raspberrypi systemd[1]: lightningd.service: Consumed 4.910s CPU time.
Apr 01 06:23:43 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2804.
Apr 01 06:23:43 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
Apr 01 06:23:43 raspberrypi systemd[1]: lightningd.service: Consumed 4.910s CPU time.
Apr 01 06:23:43 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
Apr 01 06:23:43 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
Apr 01 06:24:04 raspberrypi sh[843302]: lightningd: /home/bitcoin/.lightning/config line 20: watchtower-proxy=127.0.0.1:9050: unrecognized option
Apr 01 06:24:04 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 06:24:04 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
Apr 01 06:24:04 raspberrypi systemd[1]: lightningd.service: Consumed 4.896s CPU time.
Apr 01 06:24:34 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2805.
[... cut ...]
Apr 01 06:27:33 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
Apr 01 06:27:33 raspberrypi systemd[1]: lightningd.service: Consumed 5.060s CPU time   

lightning-cli: Connecting to 'lightning-rpc': Connection refused
/home/admin/BBfundWallet.sh: line 20: [: 783417: unary operator expected
# FAIL PRECHECK - 'sudo -u bitcoin /usr/local/bin/lightning-cli --conf=/home/bitcoin/.lightning/config getinfo' blockheight is different from 'bitcoind getblockchaininfo' - wait until chain is sync

# PRESS ENTER to return to menu

openoms, 2023-04-01:
```
mainnet.
Apr 01 06:25:49 raspberrypi sh[853865]: lightningd: /home/bitcoin/.lightning/config line 20: watchtower-proxy=127.0.0.1:9050: unrecognized option
```
delete this line from the config and restart lightningd
(or install the watchtower plugin you used before)

JoFi7, 2023-04-01:
It's worked out. Thanks very much. The Watchtower was still from the 1.9.0rc2 installation

closed 2023-04-01 by JoFi7