raspiblitz/raspiblitz #3727: Request to Support Litd from the Upgrade Menu

jbrill, 2023-03-31:
I'm hoping to upgrade litd locally to make use of autofees. Would it be possible to support an upgrade option within the upgrade menu? Thank you!

rootzoll, 2023-04-03:
Best would be to check who worked on Litd before and ask those devs to contribute more or ask the litd people if they can contribute or even sponsor a deeper integration with a bounty.

jbrill, 2023-04-04:
@rootzoll thank you for your reply. I put up a PR here https://github.com/rootzoll/raspiblitz/pull/3738 which I hope addresses this version upgrade. Note that this PR simply upgrades the install version, and does not address a UI for upgrading. I followed this [commit](https://github.com/rootzoll/raspiblitz/commit/216974b40f274cc0ec0123be6dfb5005137545f3) from @openoms on a similar LIT version upgrade PR.

rootzoll, 2023-05-03:
#3738 merged and will be part of v1.9.0 release - closing issue.

closed 2023-05-03 by rootzoll