raspiblitz/raspiblitz #3729: Use Tor socks for CoinJoin plugin BTCPAY server

kravens, 2023-04-02:
[image]
I've been playing with the new plugin from BTCPay server that allows to do CoinJoins and host a coordinator service, but I get this error that only clearnet is used. From Kukks the dev I got this specific configuration line that is missing: https://github.com/btcpayserver/btcpayserver-docker/blob/6e9b85a9c82a78b6bd5b6ddba330bd853a4998ba/docker-compose-generator/docker-fragments/opt-add-tor.yml#L14

>  btcpayserver:
    environment:
      HIDDENSERVICE_NAME: BTCPayServer
      HIDDENSERVICE_REVERSEPROXY: nginx
      BTCPAY_TORRCFILE: /usr/local/etc/tor/torrc-2
      BTCPAY_SOCKSENDPOINT: tor:9050
    volumes:
      - "tor_servicesdir:/var/lib/tor/hidden_services"
      - "tor_torrcdir:/usr/local/etc/tor/"

Is it possible to add this line by default in Raspiblitz? (I'm on 1.9 branch, running behind Tor but added hybrid clearnet for LND and BTCPay myself)
Or should I set it up manually somewhere?

kravens, 2023-04-02:
[image]
Fun fact, the BTCPay Server is actually running behind Tor while showing this error.

openoms, 2023-04-14:
Managed to reproduce, but these are settings for a Docker setup. Maybe @Kukks could give us some pointers what is expected here if BTCpay is not running in Docker and Tor is available?

Kukks, 2023-04-15:
I believe a  config line  `socksendpoint=tor:9050` should do the trick

openoms, 2023-04-15:
> socksendpoint=tor:9050

thanks! Adding:
```
socksendpoint=127.0.0.1:9050
```
to the file:
```
/home/btcpay/.btcpayserver/Main/settings.config
sudo systemctl restart btcpayserver
```
did it:
```
info: PayServer:      Connected to WebSocket of NBXplorer (BTC)
info: Configuration:  Now listening on: http://127.0.0.1:23000
info: zksnacks:       Connected to coordinator
```

Adding it to the the default config.

Kukks, 2023-04-15:
> thanks! Adding:
> ```
> socksendpoint=127.0.0.1:9050
> ```
> to the file:
> ```
> /home/btcpay/.btcpayserver/Main/settings.config
> sudo systemctl restart btcpayserver
> ```
> did it:
> ```
> info: PayServer: [...]

For future reference,  all the docker env variables are usable in the configfl file if your remove the BTCPAY_ prefix

kravens, 2023-04-16:
> For future reference, all the docker env variables are usable in the configfl file if your remove the BTCPAY_ prefix

Can confirm that adding this line fixed the configuration for me too, no more warnings that I'm coinjoining via clearnet! Thanks for the help ⚡

commit 8075ad62 (2023-04-15): add socksendpoint=127.0.0.1:9050 to fix #3729
commit 08430453 (2023-04-23): btcpay update to v1.9.1, migration to postgres (#3756)

* btcpay update to v1.9.0, migration to postgres
* update CHANGES.md
* always fix postgres files ownership
* add socksendpoint=127.0.0.1:9050 to fix #3729
* btcpay update to v1.9.1
* update changes
* create config directory in BtcPayConfig
closed 2023-04-16 by kravens