raspiblitz/raspiblitz #3730: Blockchain progress gives several restarts after minutes and stops

smartbart1, 2023-04-02:
Blockchain progress get stuck. 

" Failed to start Bitcoin daemon"

Already tried flashing ssd again.

What can I do?

Log:

`***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.8.0
chainnetwork: bitcoin / test
 16:23:08 up  3:05,  2 users,  load average: 2.67, 2.61, 2.68

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2023-04-02 14:07:46 BST; 2h 15min ago
    Process: 324163 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 324174 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=1/FAILURE)
        CPU: 62ms

Apr 02 14:07:46 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Apr 02 14:07:46 raspberrypi systemd[1]: Failed to start Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Wed 2022-07-20 12:10:49 BST, ends at Sun 2023-04-02 16:23:08 BST. --
Apr 02 14:07:46 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Apr 02 14:07:46 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Apr 02 14:07:46 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
Apr 02 14:07:46 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 6.
Apr 02 14:07:46 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Apr 02 14:07:46 raspberrypi systemd[1]: bitcoind.service: Start request repeated too quickly.
Apr 02 14:07:46 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Apr 02 14:07:46 raspberrypi systemd[1]: Failed to start Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2023-04-02T13:05:15Z UpdateTip: new best=000000000000046ca16470171307eb6e43acfa20a24632c09478a047899818a1 height=214002 version=0x00000001 log2_work=69.218606 tx=10432417 date='2012-12-28T11:10:15Z' progress=0.012825 cache=468.3MiB(3461090txo)
2023-04-02T13:05:15Z UpdateTip: new best=000000000000035062f487684d4786271810a5b891da488e5b0035f1c380bd76 height=214003 version=0x00000002 log2_work=69.218633 tx=10432878 date='2012-12-28T11:17:24Z' progress=0.012826 cache=468.3MiB(3461138txo)
2023-04-02T13:05:15Z UpdateTip: new best=000000000000039004f3b036dfe3a55bf4fa58e8eccb2a3182310e9ea2f6c944 height=214004 version=0x00000001 log2_work=69.218660 tx=10433133 date='2012-12-28T11:24:04Z' progress=0.012826 cache=468.3MiB(3461143txo)
2023-04-02T13:05:15Z UpdateTip: new best=00000000000003648c6adb0c4b458739e79c953854b1c7e2ae2206e6cd7d9252 height=214005 version=0x00000002 log2_work=69.218687 tx=10433145 date='2012-12-28T11:23:44Z' progress=0.012826 cache=468.3MiB(3461127txo)
2023-04-02T13:05:15Z UpdateTip: new best=0000000000000121a01c5f271d3f6a1fe00169b056d4158494850911877c8757 height=214006 version=0x00000001 log2_work=69.218714 tx=10433152 date='2012-12-28T11:24:26Z' progress=0.012826 cache=468.3MiB(3461118txo)
2023-04-02T13:05:15Z UpdateTip: new best=00000000000001d50183c9d4b462276f3dd67b677187a2aa9d0f4036f5cbf817 height=214007 version=0x00000001 log2_work=69.218741 tx=10433195 date='2012-12-28T11:26:36Z' progress=0.012826 cache=468.3MiB(3460998txo)
2023-04-02T13:06:37Z Unable to open file /mnt/hdd/bitcoin/blocks/blk00034.dat
2023-04-02T13:06:37Z ERROR: WriteBlockToDisk: OpenBlockFile failed
2023-04-02T13:06:37Z *** Failed to write block
2023-04-02T13:06:37Z Error: A fatal internal error occurred, see debug.log for details
2023-04-02T13:06:37Z tor: Thread interrupt
2023-04-02T13:06:37Z Shutdown: In progress...
2023-04-02T13:06:37Z addcon thread exit
2023-04-02T13:06:37Z RemoveLocal(bphqzt2bz57nthuw3kuhnhdxcjjo7gbgfrke32trrgsipq6jozw7wdyd.onion:8333)
2023-04-02T13:06:37Z torcontrol thread exit
2023-04-02T13:06:37Z opencon thread exit
2023-04-02T13:06:37Z ERROR: ProcessNewBlock: AcceptBlock FAILED (AcceptBlock: Failed to find position to write new block to disk)
2023-04-02T13:06:37Z msghand thread exit
2023-04-02T13:06:37Z DeleteAuthCookie: Unable to remove random auth cookie file: filesystem error: cannot remove: Read-only file system [/mnt/hdd/bitcoin/.cookie]
2023-04-02T13:06:37Z net thread exit
[... 200 lines cut ...]
btc_default_online="0"
btc_default_mempool_transactions="0"
btc_default_ready="0"
btc_default_error_full=""
btc_default_sync_percentage="1.23"
btc_default_synced="0"
btc_default_blocks_headers="783616"
btc_default_error_short=""
btc_default_blocks_verified="212614"
btc_default_sync_progress="0.01231570998585008"
btc_default_running="0"

*** LOGFILES ***
Archived and active journals take up 176.0M in the file system.
189M    /var/log

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

Press ENTER to return to main menu.
`

smartbart1, 2023-04-02:
Also, when trying to remove hdd completely te strart fresh it returns:

rm: cannot remove '/mnt/hdd/*': Read-only file system

rootzoll, 2023-04-03:
Yeah something with your SSD looks quite strange. Is that a selfbuild setup or a ready-build node (e.g. from Fulmo)?

smartbart1, 2023-04-04:
it's selfbuild, bought Samsung T7 Portable SSD 1TB and never used before. Reformatted ssd, but same problem. I will now try to download blockchain on windows and ssh to raspberry pi.

Btw, also didn't get my lcd elegoo working, but that may be another isseu and is not that important for me.

smartbart1, 2023-04-06:
@rootzoll 

2023-04-06T18:39:30Z Unable to open file /mnt/hdd/bitcoin/blocks/rev00007.dat
2023-04-06T18:39:30Z ERROR: UndoWriteToDisk: OpenUndoFile failed
2023-04-06T18:39:30Z *** Failed to write undo data
2023-04-06T18:39:30Z Error: A fatal internal error occurred, see debug.log for details
2023-04-06T18:39:30Z tor: Thread interrupt
2023-04-06T18:39:30Z Shutdown: In progress...
2023-04-06T18:39:30Z DeleteAuthCookie: Unable to remove random auth cookie file: filesystem error: cannot remove: Read-only file system [/mnt/hdd/bitcoin/.cookie]
2023-04-06T18:39:30Z ERROR: ConnectTip: ConnectBlock 00000000000007982b6fd9de79e64b18b356835c6865b1380b9483f9ce4b15b4 failed, Failed to write undo data
2023-04-06T18:39:30Z opencon thread exit
2023-04-06T18:39:30Z ERROR: ProcessNewBlock: ActivateBestChain failed (Failed to write undo data)
2023-04-06T18:39:30Z msghand thread exit
2023-04-06T18:39:30Z RemoveLocal(jczcfdsw2g2hujxsj5cnvqqw7cbwdp3vm5owhbsrmsgaen3rdpoeucad.onion:8333)
2023-04-06T18:39:30Z torcontrol thread exit
2023-04-06T18:39:30Z addcon thread exit
2023-04-06T18:39:30Z net thread exit

cricktor, 2023-04-19:
A Samsung T7 might be too power hungry, while the Samsung T5 seems to work fine. What are the specs of the T7 regarding power consumption?

Inspect your logs with respect to low power issues detected because all the USB interfaces share a common max. power budget of not more than 1.2A on a Raspi 4B. Do you use the true original Raspi power supply? Bad clones could be a nightmare.

closed 2024-08-10 by openoms